YosysHQ/yosys #707: yosys install on linux centos   (yosys-yosys-0.8)

vlsimonks, 2018-11-22:
Hi , 

getting below error while installing yosys in centos , can you please here .

[root@Vlsimonks yosys-yosys-0.8]# make
[Makefile.conf] CONFIG := gcc
[  0%] Building kernel/version_UNKNOWN.cc
[  0%] Building kernel/version_UNKNOWN.o
[  0%] Building kernel/driver.o
[  0%] Building techlibs/common/simlib_help.inc
[  0%] Building techlibs/common/simcells_help.inc
[  1%] Building kernel/register.o
[  1%] Building kernel/rtlil.o
[  2%] Building kernel/log.o
[  2%] Building kernel/calc.o
[  3%] Building kernel/yosys.o
[  3%] Building kernel/cellaigs.o
[  4%] Building kernel/celledges.o
[  4%] Building libs/bigint/BigIntegerAlgorithms.o
[  5%] Building libs/bigint/BigInteger.o
[  5%] Building libs/bigint/BigIntegerUtils.o
[  6%] Building libs/bigint/BigUnsigned.o
[  6%] Building libs/bigint/BigUnsignedInABase.o
[  7%] Building libs/sha1/sha1.o
[  7%] Building libs/subcircuit/subcircuit.o
[  8%] Building libs/ezsat/ezsat.o
[  8%] Building libs/ezsat/ezminisat.o
[  9%] Building libs/minisat/Options.o
[  9%] Building libs/minisat/SimpSolver.o
[  9%] Building libs/minisat/Solver.o
[ 10%] Building libs/minisat/System.o
[ 10%] Building frontends/ast/ast.o
[ 11%] Building frontends/ast/simplify.o
[ 11%] Building frontends/ast/genrtlil.o
[ 12%] Building frontends/ast/dpicall.o
[ 12%] Building frontends/blif/blifparse.o
[ 12%] Building frontends/ilang/ilang_parser.tab.cc
[ 13%] Building frontends/ilang/ilang_parser.tab.o
[ 13%] Building frontends/ilang/ilang_lexer.cc
[ 13%] Building frontends/ilang/ilang_lexer.o
[ 14%] Building frontends/ilang/ilang_frontend.o
[ 14%] Building frontends/json/jsonparse.o
[ 15%] Building frontends/liberty/liberty.o
[ 15%] Building frontends/verific/verific.o
frontends/verific/verific.cc:1679:6: warning: ‘bool {anonymous}::check_noverific_env()’ defined but not used [-Wunused-function]
 bool check_noverific_env()
      ^~~~~~~~~~~~~~~~~~~
[ 15%] Building frontends/verilog/verilog_parser.tab.cc
frontends/verilog/verilog_parser.y:142.21-27: syntax error, unexpected identifier
make: *** [frontends/verilog/verilog_parser.tab.cc] Error 1
[root@Vlsimonks yosys-yosys-0.8]# p

Thanks,
Vijay

daveshah1, 2018-11-24:
The problem is that certain CentOS versions come with an archeological version of Bison that doesn't support the syntax Yosys uses to provide higher quality error messages. See #332 which is a duplicate of this.

closed 2018-11-24 by daveshah1

B131884, 2019-12-12:
Then, can you please provide us the proper installation commands and all packages which are supported in centos?