YosysHQ/yosys #710: Yosys fails to infer BRAM with complex enable logic

daveshah1, 2018-11-28:
## Steps to reproduce the issue

Run `synth_ice40` on the below design:

```
module top(input clk, input we, re, input [7:0] addr, wdata, output reg [7:0] rdata);

reg [7:0] bram[0:255];
(* keep *) reg dummy;

always @(posedge clk)
	if (reset)
		dummy <= 1'b0;
	else if (re)
		rdata <= bram[addr];
	else if (we)
		bram[addr] <= wdata;
endmodule

```

## Expected behavior

BRAM is inferred, with both read and write enables ANDed with !reset

## Actual behavior

Yosys fails to infer BRAM with the following message:

```
    Mapping to bram type $__ICE40_RAM4K_M0 (variant 1):
      Write port #0 is in clock domain \clk.
        Mapped to bram port B1.
      Read port #0 is in clock domain !~async~.
        Bram port A1.1 has incompatible clock type.
        Failed to map read port #0.
    Mapping to bram type $__ICE40_RAM4K_M0 failed.
```
This is because no flipflop is merged into the read port:
```
Checking cell `$memrd$\bram$read_enable.v:10$6' in module `\top': no (compatible) $dff found.
```

daveshah1, 2018-11-28:
cc @mattvenn whose design this was reduced from

br1985, 2019-04-29:
+1

Any known workaround? I'm also experiencing very similar issue...

BTW, this may be related: https://github.com/YosysHQ/yosys/issues/335

PR #1130 (2019-06-25): memory_dff: walk through more than one mux for computing read enable
Fixes #710
closed 2019-06-25 by cliffordwolf