YosysHQ/yosys #728: equiv_make is confused by feedback through module port

whitequark, 2018-12-07:
## Steps to reproduce the issue

Run the following script:
```
read_ilang bug.il
equiv_make gold gate equiv
hierarchy -top equiv
techmap -autoproc -map +/simcells.v
equiv_induct
equiv_status -assert
```
<details>
<summary>bug.il</summary>

```verilog
module \gold
  wire output 2 \o
  wire input 1 \sys_clk
  attribute \init 3'001
  wire \v
  wire \v$next
  wire input 3 \x
  wire input 4 \y
  cell $_MUX_ \_0_
    connect \A \o
    connect \B \x
    connect \S \y
    connect \Y \v$next
  end
  cell $_DFF_P_ \_1_
    connect \C \sys_clk
    connect \D \v$next
    connect \Q \o
  end
  connect \v \o
end
module \gate
  wire $auto$dff2dffe.cc:158:make_patterns_logic$5123
  wire output 2 \o
  wire input 1 \sys_clk
  attribute \init 3'001
  wire \v
  wire \v$next
  wire input 3 \x
  wire input 4 \y
  cell $_XOR_ $auto$simplemap.cc:85:simplemap_bitop$5127
    connect \A \y
    connect \B 1'0
    connect \Y $auto$dff2dffe.cc:158:make_patterns_logic$5123
  end
  cell $_MUX_ \_0_
    connect \A 1'x
    connect \B \x
    connect \S \y
    connect \Y \v$next
  end
  cell $_DFFE_PP_ \_1_
    connect \C \sys_clk
    connect \D \v$next
    connect \E $auto$dff2dffe.cc:158:make_patterns_logic$5123
    connect \Q \o
  end
  connect \v \o
end
```
</details>

The `gate` module was produced from `gold` module by running `dff2dffe -direct-match $_DFF_*`.

## Expected behavior

Equivalence check succeeds.

## Actual behavior

Equivalence check fails.

My understanding of this is as follows. Here are the `gold` and `gate` modules:

[image]
[image]

When combined, it looks like `equiv_make` is confused by module ports, and wires one of the `$equiv` cells as an input to another `$equiv` cell via a combinatorial network. This probably breaks it:

[image]

whitequark, 2019-01-07:
@cliffordwolf Do you think you could take a look into it? I don't understand the `equiv_make` pass nearly well enough, and false positives/negatives with it are rather costly when debugging passes elsewhere...

cliffordwolf, 2019-03-03:
Adding a `rename` command to hide things such as the name of the `$MUX` cell makes it pass:

```
read_ilang bug.il
rename -hide n:_*_ n:*$*
equiv_make gold gate equiv
hierarchy -top equiv
techmap -autoproc -map +/simcells.v
equiv_induct
equiv_status -assert
```

Otherwise `equiv_make` will assume that the mux `_0_` in gold and the mux `_0_` in gate are equivalent because they have the same name. But they are not equivalent, because `dff2dffe` re-wires the A input of that mux. (Same for `v$next`: the two nets of that name are not equivalent after `dff2dffe`.)

closed 2019-03-03 by cliffordwolf

whitequark, 2019-03-03:
@cliffordwolf Thanks for the investigation! However, wouldn't it be better to teach `equiv_make` that cells of the identical name but different type should not be considered identical?

cliffordwolf, 2019-03-03:
That's already what's happening. If they would be of different types they would not be considered equivalent. Same if they are of the same type but with different parameters. (See `find_same_cells()` in `equiv_make.cc`.) But those cells are both of type `$_MUX_`.

whitequark, 2019-04-24:
@cliffordwolf I'd like to make this work out of the box for the `equiv_opt` pass. What do you think is the best way to do it, add the `rename -hide` call?

cliffordwolf, 2019-04-28:
I don't think you can do anything here "out of the box" because the whole idea of `equiv_make` is that it is allowed to match things of the same name.

You could add some extra rules, like say don't use names matching regexes like `_[0-9]+_`, and an extra option that disables those extra rules.

Maybe even add a `-whitelist` option that allows to specify the names that `equiv_make` is allowed to use.

But if you just hide _all_ the names indiscriminately then you are just crippling `equiv_make`.

Another question is if `dff2dffe` should do anything to mark the paths it makes non-equivalent so that a later `equiv_make` doesn't struggle with it. But that needs to be done in `dff2dffe`. Afaict there is no easy way for `equiv_make` to know which paths are effected by this.

(People using external equivalence checkers, such as formality, would run into the exact same problems. There you just manually blacklist names of problematic nodes until the check succeeds.)

whitequark, 2019-04-28:
Thanks. That was my guess but I wanted to recheck. If someone (including me) hits this problem again we can revisit this.