YosysHQ/yosys #743: fatal error: invalid formal-actual when running arachne-pnr

adumont, 2018-12-16:
## Steps to reproduce the issue

My project hasn't changed, but with latest yosys the automated synthesis fails. Yosys has changed (arachne-pnr hasn't).

Project source: https://github.com/adumont/hrm-cpu/commit/ba354e9f63cba8663fe1883d4a2a4d0831743900

Good: https://travis-ci.org/adumont/hrm-cpu/builds/465638448
Toolchain sha1:
/home/travis/toolchain/arachne-pnr.ver:840bdfdeb38809f9f6af4d89dd7b22959b176fdd
/home/travis/toolchain/icestorm.ver:9671b760f84ca4006f0ef101a3e3b201df4eabb5
/home/travis/toolchain/symbiyosys.ver:970e4b2857103d5237d6069e4e753b978dc480c4
/home/travis/toolchain/yices2.ver:35e39612c7402a42bac95aa24b3484be6d31200e
/home/travis/toolchain/yosys.ver:**47a5dfdaa4bd7d400c6e3d58476de80904df460d**

Bad: https://travis-ci.org/adumont/hrm-cpu/builds/468681334
/home/travis/toolchain/arachne-pnr.ver:840bdfdeb38809f9f6af4d89dd7b22959b176fdd
/home/travis/toolchain/icestorm.ver:9671b760f84ca4006f0ef101a3e3b201df4eabb5
/home/travis/toolchain/symbiyosys.ver:0772456a15b5a474c3c90a06b48e46a67d8a6614
/home/travis/toolchain/yices2.ver:0c2b13f0696e5e4d71dcccb67e4a2e5804745002
/home/travis/toolchain/yosys.ver:**5fa5dbbddab03af89e00426e21df9f30d020a94e**

*Provide instructions for reproducing the issue. Make sure to include
all necessary source files. (You can simply drag&drop a .zip file into
the issue editor.)*

## Expected behavior

Should end with:
write_txt builddir/alhambra/top.pnr...

## Actual behavior

*Please describe how the behavior you see differs from the expected behavior.*

**Important Note:** Nobody will be able to help you and/or fix the issue if you
do not provide sufficient information for reproducing the problem.

```
arachne-pnr -d 1k -p top-alhambra.pcf builddir/alhambra/top.blif -o builddir/alhambra/top.pnr
seed: 1
device: 1k
read_chipdb +/share/arachne-pnr/chipdb-1k.bin...
  supported packages: cb121, cb132, cb81, cm121, cm36, cm49, cm81, qn84, swg16tr, tq144, vq100
read_blif builddir/alhambra/top.blif...
builddir/alhambra/top.blif:30: fatal error: invalid formal-actual
```

bad blif file: http://termbin.com/lzt31
good blif file (expected): http://termbin.com/0yx0
pcf file (just in case): http://termbin.com/0svi

adumont, 2018-12-16:
See https://freenode.irclog.whitequark.org/yosys/2018-12-16#23697653;

```
18:28 <daveshah> I'll get clifford to revert the commit that broke that 
18:30 <daveshah> A change I made occasionally adds spaces to names which breaks blif 
```

daveshah1, 2018-12-16:
Can you check the regression is fixed by #745 which should revert this change?

adumont, 2018-12-16:
Yes, #745 fixes this issue for me:

https://travis-ci.org/adumont/hrm-cpu/builds/468681334 Build is successful
:)

Thanks a lot

closed 2018-12-17 by cliffordwolf