YosysHQ/yosys #756: Cannot synthesis wb_common from FuseSoC

Icenowy, 2018-12-20:
## Steps to reproduce the issue

Let Yosys run read_verilog on the wb_common verilog file from wb_common.
https://github.com/fusesoc/wb_common/blob/master/wb_common.v

## Expected behavior

The file should be successfully processed.

## Actual behavior

```

ZipCPU, 2019-01-18:
The problem involved stems from the fact that $clog2 only works with compile time constants.  These can be generated with literals, such as 25 or 32'hdead_beef, or from parameters.  Further, if the compile time constant comes from a parameter, that parameter needs to have a default definition in order to avoid this error.

We would like to implement this featur

cliffordwolf, 2019-01-18:
From the Verilog Synthesis standard (IEEE Std 1364.1):

[image]

Yosys, however, does support `$clog2` (and a few other system functions), but only on simple compile time constants.

ZipCPU, 2019-01-18:
@olofk -- please see above

olofk, 2019-01-18:
Released version v1.0.3 with this fixed

udif, 2019-01-19:
1. If you thing only 32 bits are used then simply support only the 32bit case, and add an assertion to fail the other cases. Do you even support runtime switching of the bus size? If not, then this should have been a parameter in the first place.
2. @ZipCPU , if you check the verilog BNF, you will see that verilog parameters must have a default value, at least for the ANSI type module declaration.

ZipCPU, 2019-02-22:
@udif : Thank you so much for that comment!  I've been dealing with a couple pieces of (somebody else's) code that haven't had defaults for their parameters.  Now I can report back to them that these reflect actual bugs in their designs.

@cliffordwolf : I think we can close this issue now.

closed 2019-02-27 by eddiehung