YosysHQ/yosys #774: yosys segfaults during MEMORY_MAP pass (on large design)

gsomlo, 2019-01-02:
## Steps to reproduce the issue
Run `yosys -o foo.blif -S ./yosys_rocket/*.v` (zipped rocket chip generated verilog here: [yosys_rocket.zip](https://github.com/YosysHQ/yosys/files/2721887/yosys_rocket.zip)).

## Expected behavior

Successful generation of `foo.blif` or some relevant error message, generated intentionally.

## Actual behavior

Segfault in "7.17. Executing MEMORY_MAP pass":
```
[...] // lots of lines removed

7.16.10. Finished fast OPT passes.

7.17. Executing MEMORY_MAP pass (converting $mem cells to logic and flip-flops).
Mapping memory cell \_T_1161 in module \BTB:
  created 512 $dff cells and 0 static cells of width 1.
  read interface: 0 $dff and 511 $mux cells.
  write interface: 512 write mux blocks.

[...] // 647 lines removed

Mapping memory cell \ram in module \data_arrays_0_ext:
  created 512 $dff cells and 0 static cells of width 512.
  read interface: 1 $dff and 511 $mux cells.
  write interface: 8192 write mux blocks.
Mapping memory cell \ram in module \mem_0_ext:
  created 512 $dff cells and 0 static cells of width 64.
  read interface: 1 $dff and 511 $mux cells.
  write interface: 4096 write mux blocks.
Mapping memory cell \ram in module \mem_ext:
Segmentation fault (core dumped)
```

## Additional Info

Stack trace from GDB:
```
#0  0x0000556a0e8389da _ZSt10_ConstructIN5Yosys5RTLIL6SigBitEJRKS2_EEvPT_DpOT0_ (yosys)
#1  0x0000556a0e83ced8 _ZNSt6vectorIN5Yosys5RTLIL6SigBitESaIS2_EE19_M_range_initializeIN9__gnu_cxx17__normal_iteratorIPKS2_S4_EEEEvT_SB_St20forward_iterator_tag (yosys)
#2  0x0000556a0ea00562 handle_cell (yosys)
#3  0x0000556a0ea042f5 MemoryMapWorker (yosys)
#4  0x0000556a0e82c86f _ZN5Yosys4Pass4callEPNS_5RTLIL6DesignESt6vectorINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESaISA_EE (yosys)
#5  0x0000556a0e82ccd5 _ZN5Yosys4Pass4callEPNS_5RTLIL6DesignENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE (yosys)
#6  0x0000556a0e82ce4c _ZN5Yosys10ScriptPass3runENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES6_ (yosys)
#7  0x0000556a0ebbf5be script (yosys)
#8  0x0000556a0ebbedc9 execute (yosys) 
#9  0x0000556a0e82c86f _ZN5Yosys4Pass4callEPNS_5RTLIL6DesignESt6vectorINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESaISA_EE (yosys)
#10 0x0000556a0e82ccd5 _ZN5Yosys4Pass4callEPNS_5RTLIL6DesignENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE (yosys)
#11 0x0000556a0e87731a _ZN5Yosys8run_passENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEPNS_5RTLIL6DesignE (yosys)
#12 0x0000556a0e811584 main (yosys)
#13 0x00007f636a1f411b __libc_start_main (libc.so.6)
#14 0x0000556a0e81ff0a _start (yosys)
```
Also, the bt output from GDB:
```
#0  std::_Construct<Yosys::RTLIL::SigBit, Yosys::RTLIL::SigBit const&> (
    __p=0x556a31155c10) at /usr/include/c++/8/new:169
#1  std::__uninitialized_copy<false>::__uninit_copy<__gnu_cxx::__normal_iterator<Yosys::RTLIL::SigBit const*, std::vector<Yosys::RTLIL::SigBit, std::           allocator<Yosys::RTLIL::SigBit> > >, Yosys::RTLIL::SigBit*> (
    __result=__result@entry=0x556a31155c10,
    __last=<error reading variable: Cannot access memory at address 0x400>,
    __first=non-dereferenceable iterator for std::vector)
    at /usr/include/c++/8/bits/stl_uninitialized.h:83
#2  std::uninitialized_copy<__gnu_cxx::__normal_iterator<Yosys::RTLIL::SigBit const*, std::vector<Yosys::RTLIL::SigBit, std::allocator<Yosys::RTLIL::SigBit> >  >, Yosys::RTLIL::SigBit*> (__result=__result@entry=0x556a31155c10,
    __last=<error reading variable: Cannot access memory at address 0x400>,
    __first=non-dereferenceable iterator for std::vector)
    at /usr/include/c++/8/bits/stl_uninitialized.h:134
#3  std::__uninitialized_copy_a<__gnu_cxx::__normal_iterator<Yosys::RTLIL::SigBit const*, std::vector<Yosys::RTLIL::SigBit, std::allocator<Yosys::RTLIL::       SigBit> > >, Yosys::RTLIL::SigBit*, Yosys::RTLIL::SigBit> (
    __first=__first@entry=non-dereferenceable iterator for std::vector,
    __last=__last@entry=<error reading variable: Cannot access memory at address 0x400>, __result=__result@entry=0x556a31155c10)
    at /usr/include/c++/8/bits/stl_uninitialized.h:289
#4  0x0000556a0e83ced8 in std::vector<Yosys::RTLIL::SigBit, std::allocator<Yosys::RTLIL::SigBit> >::_M_range_initialize<__gnu_cxx::__normal_iterator<Yosys::    RTLIL::SigBit const*, std::vector<Yosys::RTLIL::SigBit, std::allocator<Yosys::RTLIL::SigBit> > > > (__last=..., __first=..., this=0x7ffd9c22f810)
    at kernel/rtlil.cc:3152
#5  std::vector<Yosys::RTLIL::SigBit, std::allocator<Yosys::RTLIL::SigBit> >::_M_initialize_dispatch<__gnu_cxx::__normal_iterator<Yosys::RTLIL::SigBit const*,  std::vector<Yosys::RTLIL::SigBit, std::allocator<Yosys::RTLIL::SigBit> > > > (
    __last=<error reading variable: Cannot access memory at address 0x400>,
    __first=non-dereferenceable iterator for std::vector, this=0x7ffd9c22f810)
    at /usr/include/c++/8/bits/stl_vector.h:1437
#6  std::vector<Yosys::RTLIL::SigBit, std::allocator<Yosys::RTLIL::SigBit> >::vector<__gnu_cxx::__normal_iterator<Yosys::RTLIL::SigBit const*, std::            vector<Yosys::RTLIL::SigBit, std::allocator<Yosys::RTLIL::SigBit> > >, void> (__a=...,
    __last=<error reading variable: Cannot access memory at address 0x400>,
    __first=non-dereferenceable iterator for std::vector, this=0x7ffd9c22f810)
    at /usr/include/c++/8/bits/stl_vector.h:546
#7  Yosys::RTLIL::SigSpec::extract (this=this@entry=0x7ffd9c22f9d8,
    offset=<optimized out>, length=length@entry=64) at kernel/rtlil.cc:3152
#8  0x0000556a0ea00562 in (anonymous namespace)::MemoryMapWorker::handle_cell (
    this=this@entry=0x7ffd9c22fe28, cell=<optimized out>)
    at passes/memory/memory_map.cc:177
#9  0x0000556a0ea042f5 in (anonymous namespace)::MemoryMapWorker::MemoryMapWorker (module=0x556a1041b5e0, design=0x556a10044180, this=0x7ffd9c22fe28)
    at passes/memory/memory_map.cc:349
#10 (anonymous namespace)::MemoryMapPass::execute (
    this=this@entry=0x556a0f0af220 <(anonymous namespace)::MemoryMapPass>,
    args=std::vector of length 1, capacity 1 = {...},
    design=design@entry=0x556a10044180) at passes/memory/memory_map.cc:369
#11 0x0000556a0e82c86f in Yosys::Pass::call (
    design=design@entry=0x556a10044180,
    args=std::vector of length 1, capacity 1 = {...}) at kernel/register.cc:232
#12 0x0000556a0e82ccd5 in Yosys::Pass::call (design=0x556a10044180,
    command="memory_map") at kernel/register.cc:212
#13 0x0000556a0e82ce4c in Yosys::ScriptPass::run (
    this=this@entry=0x556a0f0b2f40 <(anonymous namespace)::SynthPass>,
    command="memory_map", info="") at kernel/register.cc:320
#14 0x0000556a0ebbf5be in (anonymous namespace)::SynthPass::script (
    this=0x556a0f0b2f40 <(anonymous namespace)::SynthPass>)
    at /usr/include/c++/8/bits/char_traits.h:285
#15 0x0000556a0ebbedc9 in (anonymous namespace)::SynthPass::execute (
    this=this@entry=0x556a0f0b2f40 <(anonymous namespace)::SynthPass>,
    args=std::vector of length 1, capacity 1 = {...},
    design=design@entry=0x556a10044180) at techlibs/common/synth.cc:172
#16 0x0000556a0e82c86f in Yosys::Pass::call (
    design=design@entry=0x556a10044180,
    args=std::vector of length 1, capacity 1 = {...}) at kernel/register.cc:232
#17 0x0000556a0e82ccd5 in Yosys::Pass::call (
    design=design@entry=0x556a10044180, command="synth")
    at kernel/register.cc:212
#18 0x0000556a0e87731a in Yosys::run_pass (command="synth",
    design=0x556a10044180) at kernel/yosys.cc:948
#19 0x0000556a0e811584 in main (argc=10, argv=0x7ffd9c2309e8)
    at kernel/driver.cc:492
```

daveshah1, 2019-01-05:
Looking at the design,
```
  reg [63:0] ram [134217727:0];
```
This is attempting to create a 128M x 64 RAM; and `synth` without any RAM mapping will try and convert that RAM to gates and flipflops. This is a minimum of 8 billion flip flops and 10s-100s of billions of gates. Given each of these must store at minimum a name; cell type; as well as wires to connect them; the required memory to store all of this will be in the order of terabytes (and istr Yosys needs an OS with an ILP64 ABI to deal with more than about 3 billion hash table entries). I think it's fair to say that to say generating a BLIF is not likely to happen. But I agree a segfault isn't ideal either.

gsomlo, 2019-01-05:
In all fairness, this was my first naive attempt to throw the kitchen sink at yosys and see what happens. When instead I do something like
```
read_verilog yosys_rocket/*.v
synth_ecp5 -blif foo.blif
```
it appears to work fine (probably because it tech-maps all that RAM onto ecp5 BRAMs instead of trying to build it all out of flip-flops). I'll probably have to dig deeper into the SymbiFlow "fork" to progress  beyond that.
This issue is mostly just about telling a user they're being stupid and exiting with an error message, rather than dying in the attempt to satisfy their impossible request :) I'm not yet familiar enough with the code base to have a clue where such an error message ought to be generated, though.

daveshah1, 2019-01-05:
Out of curiosity, what architecture are you targeting? Although it does manage to map this design to ECP5 BRAM, 1GiB of RAM is 3 orders of magnitude more than the largest ECP5 has. You will need to replace that memory module with an external DRAM (SDRAM/DDR/DDR2/DDR3/etc) interface if you want to use this much memory.

gsomlo, 2019-01-05:
It's the chisel-generated verilog that goes with lowRISC (which also
adds their own DRAM controller to connect to the nexys4-ddr board's
memory. I didn't realize if I left all that out and just threw the
rocket-chip subset at yosys that it would try to synthesize the
"missing" RAM !

Guess I should try to build the *whole* thing (like I'm currently
doing with Vivado). The ultimate goal would be to port lowRISC (or
some 32-bit "smaller" version thereof) to ECP5, using yosys, trellis,
and symbiflow.

I'm in the process of getting familiar with the FOSS tools, and have
some cycles to contribute code to them (once I get over the learning
curve, hopefully soon :)

daveshah1, 2019-01-05:
This sounds like an awesome project! Incidentally, I have getting a DDR2/3 controller to work with the Yosys/nextpnr/Trellis flow for the ECP5 on my TODO list in the near future (it's part of my master's project to improve the ECP5 flow). In the mean time, it is already possible to use an SDRAM controller in this flow - I've had success both with [wb_sdram_ctrl](https://github.com/skristiansson/wb_sdram_ctrl/blob/master/rtl/verilog/sdram_ctrl.v) and [litedram](https://github.com/enjoy-digital/litedram/tree/master/litedram) for medium-size memories (32MB or 64MB, just enough to boot Linux).

I see you mentioned the SymbiFlow fork of Yosys earlier - the open source ECP5 flow is only tangentially connected to the SymbiFlow project, so you'll want to be using Yosys master where all ECP5 development is happening and not the SymbiFlow fork that might lag behind.

cliffordwolf, 2019-03-02:
Looks like the issue was resolved (OP tried to map 8 GB of ram to FFs and gates). -> closing the issue.

closed 2019-03-02 by cliffordwolf