YosysHQ/yosys #778: ECP5: reg has an unprocessed 'init' attribute

C-Elegans, 2019-01-04:
When synthesizing a design for ECP5 that has an initialization for a register, such as

    reg [30:0] 	 counter = 31'hA5A5A5A5;

Yosys emits this warning:

    Warning: Wire blinky.counter has an unprocessed 'init' attribute.

From reading the ECP5 FPGA Libraries Reference Guide, it appears that the ECP5 doesn't have a way to initialize the registers to a value. Could the solution used on iCE40 of adding inverters to the register also be implemented for ECP5?

daveshah1, 2019-01-04:
I am unsure what the best way forward is wrt ECP5 register initialisation. Unlike iCE40, ECP5 registers are not zero-initialised - in hardware GSR is effectively OR'd with LSR; so they are initialised to their reset value. This means the inverter trick is not useful.

The best option for now will probably be to process initialisation values only where a set/reset signal doesn't exist and give a warning otherwise (after all, initialisation values aren't strictly speaking synthesisable in the first place).

Clearly where async SR conflicts with the init value, this is a conflict that cannot be resolved so the only option is to error out; or ignore the init value. Conflicts between sync SR and init value are more of a grey area. In theory it would also be possible to blast the sync SR back to an AND/OR gate in front of the D input, in order to respect the init value. afaik Diamond doesn't do this; but Yosys is probably a bit more keen to infer sync SRs in the first place.

C-Elegans, 2019-01-07:
So essentially on lattice, the only consistent way of doing it is using a reset block like you would with an ASIC. I can live with that

closed 2019-02-28 by eddiehung

daveshah1, 2019-02-28:
549b8e7 fixes ECP5 FF init wherever possible:

 - init values are trivially honoured on FFs without reset, simply by choosing set or reset (and leaving the LSR input unused) as appropriate
 - init values on FFs with sync set/reset will do nothing (and be honoured) if they match the set/reset, otherwise the sync set/reset will be converted back to logic in order to honour the init value
 - init values on FFs with async set/reset will be a warning if they don't match the async LSR value