YosysHQ/yosys #781: ERROR: Found error in internal cell

tomtor, 2019-01-06:
## Steps to reproduce the issue

[test.zip](https://github.com/YosysHQ/yosys/files/2730336/test.zip)

extract `test.v` from `test.zip`

    yosys -p "synth_ice40 -blif test.blif" test.v

## Expected behavior

No error.

This design works as expected with Vivado

## Actual behavior

    ERROR: Found error in internal cell \test_deflate_bench.$auto$share.cc:714:make_supercell$43901 
    ($mux) at kernel/rtlil.cc:717:
      cell $mux $auto$share.cc:714:make_supercell$43901
        parameter \WIDTH 9
        connect \Y $auto$rtlil.cc:1754:Mux$43902
        connect \S $auto$share.cc:982:make_cell_activation_logic$43898
        connect \B \deflate0_b1
        connect \A $add$test.v:1514$1817_Y [8:0]
      end

whitequark, 2019-01-07:
I've [minimized](https://github.com/YosysHQ/yosys/files/2731146/test-minimized.zip) this testcase with [bugpoint](https://github.com/YosysHQ/yosys/pull/783) using the following script:

    share

and the following bugpoint invocation:

    bugpoint -script share.ys -grep \test_deflate_bench.$auto$share.cc:714:make_supercell$ -fast

on RTLIL that was acquired right before the `share` pass in `synth -run coarse`.

It looks related to sharing of memory ports:
<details>

```
2. Executing SHARE pass (SAT-based resource sharing).
Found 4 cells in module test_deflate_bench that may be considered for resource sharing.
  Analyzing resource sharing options for $memrd$\deflate0_iram$test.v:2352$2068 ($memrd):
    Found 1 activation_patterns using ctrl signal 1'x.
    Found 1 candidates: $memrd$\deflate0_iram$test.v:2016$2031
    Analyzing resource sharing with $memrd$\deflate0_iram$test.v:2016$2031 ($memrd):
      Found 1 activation_patterns using ctrl signal 1'x.
      Forbidden control signals for this pair of cells: { $logic_and$test.v:2352$2073_Y $logic_and$test.v:2016$2044_Y }
      Activation pattern for cell $memrd$\deflate0_iram$test.v:2352$2068: 1'x = 1'1
      Activation pattern for cell $memrd$\deflate0_iram$test.v:2016$2031: 1'x = 1'1
      According to the SAT solver the cell $memrd$\deflate0_iram$test.v:2352$2068 is never active. Sharing is pointless, we simply remove it.
  Analyzing resource sharing options for $memrd$\deflate0_iram$test.v:2016$2031 ($memrd):
    Found 1 activation_patterns using ctrl signal 1'x.
    No candidates found.
  Analyzing resource sharing options for $memrd$\deflate0_codeLength$test.v:2028$2047 ($memrd):
    Found 1 activation_patterns using ctrl signal { \reset $procmux$40818_CMP $logic_and$test.v:1995$2025_Y }.
    Found 1 candidates: $memrd$\deflate0_codeLength$test.v:1515$1818
    Analyzing resource sharing with $memrd$\deflate0_codeLength$test.v:1515$1818 ($memrd):
      Found 1 activation_patterns using ctrl signal { \reset \deflate0_do_init $procmux$40707_CMP 1'x }.
      Activation pattern for cell $memrd$\deflate0_codeLength$test.v:2028$2047: { \reset $procmux$40818_CMP $logic_and$test.v:1995$2025_Y } = 3'110
      Activation pattern for cell $memrd$\deflate0_codeLength$test.v:1515$1818: { \reset \deflate0_do_init $procmux$40707_CMP 1'x } = 4'1110
      Adding exclusive control bits: $procmux$40707_CMP vs. $procmux$40818_CMP
      Size of SAT problem: 3 cells, 248 variables, 658 clauses
      According to the SAT solver this pair of cells can be shared.
      Activation signal for $memrd$\deflate0_codeLength$test.v:2028$2047: $auto$share.cc:972:make_cell_activation_logic$1
      New cell: $auto$share.cc:710:make_supercell$3 ($memrd)
  Analyzing resource sharing options for $auto$share.cc:710:make_supercell$3 ($memrd):
    Found 2 activation_patterns using ctrl signal { \reset \deflate0_do_init $procmux$40818_CMP $procmux$40707_CMP $logic_and$test.v:1995$2025_Y 1'x }.
    No candidates found.
Removing 3 cells in module test_deflate_bench:
  Removing cell $memrd$\deflate0_codeLength$test.v:1515$1818 ($memrd).
  Removing cell $memrd$\deflate0_codeLength$test.v:2028$2047 ($memrd).
  Removing cell $memrd$\deflate0_iram$test.v:2352$2068 ($memrd).
ERROR: Found error in internal cell \test_deflate_bench.$auto$share.cc:714:make_supercell$4 ($mux) at kernel/rtlil.cc:717:
  cell $mux $auto$share.cc:714:make_supercell$4
    parameter \WIDTH 9
    connect \Y $auto$rtlil.cc:1754:Mux$5
    connect \S $auto$share.cc:972:make_cell_activation_logic$1
    connect \B \deflate0_b1
    connect \A $add$test.v:1514$1817_Y [8:0]
  end
```
</details>

cliffordwolf, 2019-01-07:
Fixed in 8a63fc5. Thanks @whitequark for the minimized test case.

closed 2019-01-07 by cliffordwolf

tomtor, 2019-01-07:
Thanks for the quick fix!