YosysHQ/yosys #784: json synthesis output should include information on port bounds

niels-moller, 2019-01-08:
This is related to issue https://github.com/YosysHQ/nextpnr/issues/206. Consider input file

    module main(// Leds, labeled D2 - D9 on the board.
    	    output [9:2] d);
    
       assign d[2] = 1;
       assign d[3] = 0;
       assign d[4] = 1;
       assign d[5] = 1;
       assign d[6] = 0;
       assign d[7] = 0;
       assign d[8] = 1;
       assign d[9] = 1;
    endmodule // main

When synthesized using 

    yosys -q -p 'synth_ice40 -json main.json' -f 'verilog' main.vl

the resulting json file seems to not include information on port bounds for [9:2], which makes it impossible for nextpnr to resolve indexing of d in the .pcf input, e.g., 

    set_io d[9] B5

niels-moller, 2019-06-16:
I'd suggest the attached patch. Not sure if the start_offset attribute fits best on "ports" or "netnames", or what the difference is. And maybe it should be omitted in the common case that it's zero.

After this patch, the section of the json output looks like

```
    "main": {
      "attributes": {
        "top": 1,
        "src": "main.vl:1"
      },
      "ports": {
        "d": {
          "direction": "output",
          "start_offset": 2,
          "bits": [ "1", "0", "1", "1", "0", "0", "1", "1" ]
        }
      },
      "cells": {
      },
      "netnames": {
        "d": {
          "hide_name": 0,
          "start_offset": 2,
          "bits": [ "1", "0", "1", "1", "0", "0", "1", "1" ],
          "attributes": {
            "src": "main.vl:2"
          }
        }
      }
    }
```  
[yosys-start_offset.txt](https://github.com/YosysHQ/yosys/files/3294107/yosys-start_offset.txt)

cliffordwolf, 2019-06-21:
#1122 added this functionality

closed 2019-06-21 by cliffordwolf