YosysHQ/yosys #785: ERROR: Assert `id2ast == NULL || mem2reg_set.count(id2ast) == 0' failed in frontends/ast/simplify.cc:3231

smunaut, 2019-01-09:
## Steps to reproduce the issue

Trying to make a small ROM and forcing yosys not to use a BRAM for that : (on an ice40)

```verilog
module f16_prom (
    input  wire [15:0] addr,
    output wire [15:0] data_out,
    input  wire clk
);
 
    (* mem2reg *) reg [15:0] rom [0:15];
    reg [15:0] do;
 
    initial
        $readmemh("bootrom.hex", rom);
 
    always @(posedge clk)
        do <= rom[addr[3:0]];
 
    assign data_out = do;
 
endmodule // f16_prom
```

## Expected behavior

No assertion and a design with 16 LUT4s.

## Actual behavior

ERROR: Assert `id2ast == NULL || mem2reg_set.count(id2ast) == 0' failed in frontends/ast/simplify.cc:3231

cliffordwolf, 2019-03-02:
Hint for next time: Include all the files needed to reproduce the issue. The contents of `bootrom.hex` are important! For example, no error with an empty `bootrom.hex` file. It's not "steps to reproduce the issue" if you don't include all the information needed to reproduce the issue. Also: The exact command you ran. (I was able to reproduce the issue in the end and am now looking into the issue.)

closed 2019-03-02 by cliffordwolf

zarem21, 2021-02-23:
How were you able to fix this error?

commit 5d93dcce (2019-03-02): Fix $readmem[hb] for mem2reg memories, fixes #785
commit 65412466 (2019-03-02): Merge pull request #847 from YosysHQ/clifford/fix785

Fix $readmem[hb] for mem2reg memories, fixes #785