YosysHQ/yosys #790: Default case statement assertions skips prior case states

ZipCPU, 2019-01-18:
## Steps to reproduce the issue

Run SymbiYosys on the following file, using the SymbiYosys configuration below:

```
module mcve(i_clk, i_value, o_value);
	input	wire		i_clk;
	input	wire	[1:0]	i_value;
	output	reg	[3:0]	o_value;

	initial	o_value = 0;
	always @(posedge i_clk)
	case(i_value)
	2'b00: o_value <= 4'h1;
	2'b01: o_value <= 4'h2;
	2'b10: o_value <= 4'h4;
	2'b11: o_value <= 4'h8;
	endcase

	always @(*)
	case(o_value)
	4'h0: begin end
	4'h1: assert(o_value == 4'h1);
	4'h2: assert(o_value == 4'h2);
	4'h4: assert(o_value == 4'h4);
	4'h8: assert(o_value == 4'h8);
	default: assert(0);
	endcase
endmodule
```

```
[options]
mode prove

[engines]
smtbmc

[script]
read -formal mcve.v
prep -top mcve

[files]
mcve.v
```

## Expected behavior

At issue is the `default: assert(0);` statement.  If the `4'h0` case statement contains only `begin end`, then the `assert(0)` will fail even the `o_value` is `4'h0`.  If there is any statement between `begin end` this behavior goes away.

This proof should pass

## Actual behavior

The proof fails, with `o_value == 0` at the `default: assert(0)` statement.

ZipCPU, 2019-01-18:
This fails with Yosys 0.8+106 (git sha1 245724a5, clang 6.0.0-1ubuntu2 -fPIC -Os)

ZipCPU, 2019-01-18:
Yosys 0.8+67 (git sha1 4357761, clang 3.8.0-2ubuntu4 -fPIC -Os) doesn't appear to have this problem

q3k, 2019-01-18:
https://github.com/YosysHQ/yosys/commit/245724a504528156485bbb32710b5e5444899ce1 looks freakishly related @whitequark

q3k, 2019-01-18:
Also fails on current HEAD (`Yosys 0.8+142 (git sha1 f3556e9f, clang 7.0.0 -fPIC -Os)`)

q3k, 2019-01-18:
Reverting 18291c20d2b17689729d9c36b08967e928e4e32a fixes the issue.

PR #793 (2019-01-19): proc_clean: fix fully def check to consider compare/signal length
Fixes #790.
commit 95b6c358 (2019-01-19): proc_clean: fix fully def check to consider compare/signal length.

Fixes #790.
closed 2019-01-19 by cliffordwolf