YosysHQ/yosys #795: MacOS: dyld: malformed mach-o image: dylib load command #12 string extends beyond end of load command

afiskon, 2019-01-22:
Hello,

I'm using yosys on MacOS Mojave 10.14.2 installed from homebrew:

```
$ brew install yosys
...
$ brew info yosys
yosys: stable 0.8 (bottled), HEAD
Framework for Verilog RTL synthesis
http://www.clifford.at/yosys/
/usr/local/Cellar/yosys/0.8_1 (97 files, 19.2MB)
  Poured from bottle on 2019-01-22 at 12:57:47
From: https://github.com/Homebrew/homebrew-core/blob/master/Formula/yosys.rb
==> Dependencies
Build: bison ✘, pkg-config ✔
Required: libffi ✔, python ✔, readline ✔
==> Options
--HEAD
	Install HEAD version
==> Analytics
install: 77 (30 days), 181 (90 days), 474 (365 days)
install_on_request: 77 (30 days), 181 (90 days), 474 (365 days)
build_error: 0 (30 days)
```

When I try to execute yosys I get an error:

```
$ yosys
dyld: malformed mach-o image: dylib load command #12 string extends beyond end of load command
Abort trap: 6
```

I'm new to MacOS thus I'm not sure what exactly causes the issue. A little peace of advice would be much appreciated.

whitequark, 2019-01-25:
Try `brew install yosys --build-from-source`.

afiskon, 2019-01-28:
@whitequark Thanks for the advice. I did the following:

```
# completely remove yosys and it's dependencies, if there are any
$ brew tap beeftornado/rmtree
$ brew rmtree yosys

$ brew upgrade
$ brew install yosys --build-from-source
$ yosys
dyld: malformed mach-o image: dylib load command #12 string extends beyond end of load command
Abort trap: 6
```

Sadly, it didn't work. 

Here are a few more weird observations:
```
$ otool -L `which yosys`
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/objdump: '/usr/local/bin/yosys': truncated or malformed object (load command 12 LC_LOAD_DYLIB library name extends past the end of the load command)

$ strings `which yosys` | egrep '\.(so|dylib)$'
/usr/lib/libc++.1.dylib
/usr/lib/libSystem.B.dylib
/usr/local/opt/readline/lib/libreadline.7.dylib
/usr/local/opt/libffi/lib/libffi.6.dylib
```

I'm trying to figure out what goes on by running `sudo ktrace trace yosys 2>&1 | gzip > trace.log.gz`  but it slows down the program **significantly**, thus it will take a while. Also I'm going to ask some friends of mine who uses MacOS how else I can debug the issue.

Meanwhile please don't hesitate to share any ideas on this issue.

whitequark, 2019-01-28:
Well... you clearly have a completely broken Yosys binary. How this happens is unfortunately beyond me.

afiskon, 2019-01-28:
OK, I manage to figure it out. Please see #801

justMaku, 2019-02-13:
What's the output of `which strip`?

afiskon, 2019-02-13:
`strip` was installed using `brew install binutils` command. 

```
$ which strip
/usr/local/opt/binutils/bin/strip

$ strip --version
GNU strip (GNU Binutils) 2.32
Copyright (C) 2019 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or (at your option) any later version.
This program has absolutely no warranty.
```

I just realized there is an alternative version `/usr/bin/strip`. Apparently it works, while the first one is broken:

```
$ strip ./yosys_a
$ /usr/bin/strip ./yosys_b
$ ./yosys_a -V
Killed: 9
$ ./yosys_b -V
Yosys 0.8+     151 (git sha1 807b3c76, clang 10.0.0 -fPIC -Os)
```

Executing:

```
$ brew rmtree binutils
$ brew install yosys
```

Seems to solve the issue.

closed 2019-03-03 by cliffordwolf

Grazfather, 2019-11-29:
+1 on cause and workaround. I don't think we should be mysteriously broken for osx users who have binutils installed. Could we skip strip on OSX specifically, or check that the version of strip is compatible?

smunaut, 2019-11-29:
Already discussed and rejected in another issue.
If you want a fix, identify why it breaks and fix that.

Grazfather, 2019-11-29:
Reported the issue to homebrew: https://github.com/Homebrew/formulae.brew.sh/issues/209, though I still think a 1 line check on darwin in the install target here would be more helpful than having everyone who uses homebrew have to debug it themselves.

Grazfather, 2019-11-29:
https://github.com/Homebrew/homebrew-core/issues/47315

baiyunping333, 2020-05-30:
in my case, because of my strip