YosysHQ/yosys #797: Verific frontend does not maintain initial values

makaimann, 2019-01-24:
## Steps to reproduce the issue

Using SymbiYosys and Yosys compiled with the Verific frontend, run:
```
unzip example.zip
cd example
sby -f run.sby pdr
```

[example.zip](https://github.com/YosysHQ/yosys/files/2793093/example.zip)

The property fails because the `initstate` register does not start with the value 1.

If you modify the `[script]` section of `run.sby` to use `read_verilog` instead of `read`, and re-run it, the property is now proven.

## Expected behavior

I would expect that the Verific frontend gives the same verification result as the internal frontend. However, I'd be happy if there were just a good way of constraining the intial state through the Verific frontend. I haven't found a way to do that yet. Perhaps there's a command for this? I looked at the yosys help but didn't see anything.

## Actual behavior

Using the Verific frontend does not preserve the initial values of registers. I also tried using an `initial` block, which Verific ignores. Thus, the Verific frontend produces a different verification result than the regular frontend.

Thank you very much for your time and please let me know if I can provide any more information!

daveshah1, 2019-01-25:
Unfortunately I can't reproduce this with the build of Yosys & Verific that I have available (it passes out of the box for me). This is probably related to the version of Verific that you are using.

cliffordwolf, 2019-01-25:
I'm assuming this with with your own build of Yosys+Verific. You need to make sure to use a version of Verific built with `DB_PRESERVE_INITIAL_VALUE` set in `<verific>/database/DBCompileFlags.h`. (See also the notes in `<yosys>/frontends/verific/README`.)

closed 2019-01-25 by cliffordwolf

makaimann, 2019-01-25:
Oops I see, sorry for bothering you guys with this then. Thanks for the advice!

mithro, 2019-01-25:
Is there a way that Yosys can check that verific has been built with the required / best set of flags?

cliffordwolf, 2019-02-24:
see a516b4f