YosysHQ/yosys #804: Can a ring-oscillator be created?

dpiegdon, 2019-02-02:
I am playing around with a paper I found [1] on generating random numbers in an FPGA.
It uses a ring oscillator and programmable delay lines to keep the generator in a metastable position.

Right now I am simply trying to synthesize a ring oscillator on an ice40lp8k (TinyFPGA BX) using yosys, but all my tries seem to have been optimized out. Sample code:

wire a;
wire b;
wire c;
assign a = ~b;
assign b = ~c;
assign c = ~a;
assign some_output_pin = c;

Obviously this is code you'd usually never want, and it yields in no gates at all (checked in floorplan).
Is there a way to get the wanted result?
I have already tried looking for corresponding yosys options to disable optimization, not use abc, and even patched the synth_ice40 command to disable any optimization.
No success so far.

Thanks!

[1] http://aceslab.org/sites/default/files/wifs2010.pdf and also
https://www.iacr.org/workshops/ches/ches2011/presentations/Session%201/CHES2011_Session1_2.pdf

dpiegdon, 2019-02-02:
Actually submitted the wrong paper link. Those are related, but here is the one I currently am working with:
https://people.csail.mit.edu/devadas/pubs/ches-fpga-random.pdf

dpiegdon, 2019-02-02:
Ok I found your example at http://svn.clifford.at/handicraft/2015/ringosc/ and this seems to properly work.

So consider this ticket obsolete. :)

closed 2019-02-02 by dpiegdon

DurandA, 2019-02-09:
Aren't inverters supposed to be initialized with 16'b0000000000000001 like below?
```
SB_LUT4 #(
	.LUT_INIT(16'd1) // instead of 16'd2
) buffers [99:0] (
	.O(buffers_out),
	.I0(buffers_in),
	.I1(1'b0),
	.I2(1'b0),
	.I3(1'b0)
);
```
[image]

dpiegdon, 2019-02-10:
the inversion is actually done not in the LUT but in general logic:  

`
    assign chain_in = !chain_out;
`

the LUT is only working as a delay line here, and to make sure the compiler is not removing the logic.
when you only use the line above, all is removed, as mentioned in the ticket description.
I also tried only using a LUT with 16'h1 as init value, but that either did not work, or the output VPP of the oscillator was so low that I could not measure it. but the logic *was* in the ASC file. I checked on the ice40-viewer and in the nextpnr-gui.

dpiegdon, 2019-02-10:
sorry, I am wrong. my oscilloscope could not see the signal, but my spectrum analyzer actually does.
i can see a -25dBm peak at roughly 625 MHz. no wonder my cheap rigol could not see it :).
but the signal is so weak, it might be useless. -25 dBm is 35.560 mVpp. so even internal logic using the oscillator may not properly work, even though it will be stronger in the internal logic.
so I think putting in a single-LUT delay is a good choice here.