YosysHQ/yosys #809: Unused wires are not removed

CapacitorSet, 2019-02-10:
Example code:

```verilog
module addop(C, O, A, B);
   input  [3:0] A;
   input  [3:0] B;
   output [3:0] O;
   output       C;
   assign {C, O} = A + B;
endmodule
```

Invocation:

    read_verilog /home/capacitorset/file.v
    hierarchy -check -top addop
    proc; opt; memory; opt; fsm; opt; techmap; opt
    abc -liberty tfhe.lib
    write_verilog yosys.out

[This](https://gist.github.com/CapacitorSet/0a1aa11cdc9dc8b3d0d17d7d2b7ec186) is the output, you can see that wires from 25 to 33 are declared but unused.

[This](https://gist.github.com/CapacitorSet/83b1d26d83aba7b3d68277bd0055cc75) is the logfile.

daveshah1, 2019-02-10:
Running `opt_clean` after ABC removes the unused wires.

CapacitorSet, 2019-02-10:
Oh well, it's that simple. Thank you.

closed 2019-02-10 by CapacitorSet