YosysHQ/yosys #813: Logic synthesis produces wrong result

CapacitorSet, 2019-02-18:
## Steps to reproduce the issue

 * Synthesize the file `add.v` (a 2-bit adder with overflow detection) with the ABC library `tfhe.lib` from https://gist.github.com/CapacitorSet/fa85bed1be4b6345b7902ae213867870.
 * Run the circuit with inputs a[0] = 1, a[1] = 0, b[0] = 1, b[0] = 0.
 * Repeat with `add_fixed.v` from the same gist.

## Expected behavior

The result of the addition is `10 + 10 = 01` (little-endian). It overflows.

The first adder reports out[0] = 0, out[1] = 1, over_or_underflow = 1, extra = 0, not_out_msb = 0.

The second adder reports out[0] = 0, out[1] = 1, over_or_underflow = 1, extra = 0, out_msb = 1.

## Actual behavior

The first adder reports out[0] = 0, out[1] = 1, **over_or_underflow = 0**, extra = 0, not_out_msb = 0.

The second adder reports out[0] = 0, out[1] = 1, over_or_underflow = 1, extra = 0, out_msb = 1.

---

`add_fixed.v` is a 2-bit adder with a small overflow-detection circuit. `add.v` is the same circuit, but with a double negation on `out[MSB]`. The latter fails to detect an overflow on `10 + 10` (little endian); furthermore, it contrasts with line 14 (`over_or_underflow = extra ^ (!not_out_msb)` evaluates to `0 ^ (!0) = 1`).

Yosys version: 0.8. Invocation:

```sh
echo "Executing Yosys..."
yosys -p "read_verilog $1; proc; opt; memory; opt; fsm; opt; techmap; opt; abc -liberty tfhe.lib; opt; write_verilog output.v"
echo "Stripping comments..."
grep -Fv '(*' output.v > stripped.v
```

daveshah1, 2019-02-18:
I am unable to reproduce with the following testbench

```
module tb();

reg [1:0] a, b;
wire [1:0] out;
wire over_or_underflow;

add add_i(.a(a), .b(b), .out(out), .over_or_underflow(over_or_underflow));

initial begin
	a = 2'b01;
	b = 2'b01;
	#5;
	$display("%b %b\n", out, over_or_underflow);
	$finish;
end

endmodule
```

and converting the liberty file to a Verilog for simulation using `yosys -p "read_liberty tfhe.lib; write_verilog tfhe.v"`

CapacitorSet, 2019-02-18:
Sorry, that was an error on my part: `MUX` is defined incorrectly (it should be `(SEL&T)+(SEL'&F)`, it was `(SEL&T)+(SEL'&F')`). Fixing that returns correct results.

closed 2019-02-18 by CapacitorSet