YosysHQ/yosys #815: ERROR: Can't open ABC output file

balanx, 2019-02-20:
## Steps to reproduce the issue

test file
```
module test (input [31:0] i, output reg y);

always @* begin
  case (i)
    3 : y = 1'b1;
    4 : y = 1'b0;
    default: y = 1'bx;
  endcase
end

endmodule
```

synth_gates.ys
```
#read_verilog test.v

hierarchy -top test
proc; flatten

synth

dfflibmap -prepare -liberty ./script/synth_gates.lib
abc -dff -liberty ./script/synth_gates.lib
dfflibmap -liberty ./script/synth_gates.lib

stat
show
write_blif test.blif
```

command
```
C:\> yosys  test.v  synth_gates.ys
```

synth_gates.lib on
‘https://github.com/cliffordwolf/picorv32/blob/master/scripts/yosys/synth_gates.lib’

## Expected behavior

can I specify '\<abc-temp-dir\>' position ?

## Actual behavior

ERROR: Can't open ABC output file `C:\Users\***\AppData\Local\Temp\\yosys-abc-ALJrvg/output.blif'

eddiehung, 2019-02-27:
Hi @balanx. I can reproduce your error (with a few more lines of context):

```
<<SNIP>>
7.2.1. Executing ABC.
Running ABC command: <yosys-exe-dir>/yosys-abc -s -f <abc-temp-dir>/abc.script 2>&1
ABC: ABC command line: "source <abc-temp-dir>/abc.script".
ABC:
ABC: + read_blif <abc-temp-dir>/input.blif
ABC: + read_lib -w <yosys-exe-dir>/synth_gates.lib
ABC: Parsing finished successfully.  Parsing time =     0.00 sec
ABC: Warning: Templates are not defined.
ABC: Libery parser cannot read "time_unit".  Assuming   time_unit : "1ns".
ABC: Libery parser cannot read "capacitive_load_unit". Assuming   capacitive_load_unit(1, pf).
ABC: Scl_LibertyReadGenlib() skipped sequential cell "DFF".
ABC: Library "gates" from "<yosys-exe-dir>/synth_gates.lib" has 3 cells (1 skipped: 1 seq; 0 tri-state; 0 no func; 0 dont_use).  Time =     0.00 sec
ABC: Memory =    0.00 MB. Time =     0.00 sec
ABC: Warnings: genlib library reader cannot detect the buffer gate.
ABC: Some parts of the supergate-based technology mapper may not work correctly.
ABC: + strash
ABC: + ifraig
ABC: + scorr
ABC: Warning: The network is combinational (run "fraig" or "fraig_sweep").
ABC: + dc2
ABC: + dretime
ABC: + strash
ABC: + &get -n
ABC: + &dch -f
ABC: + &nf
ABC: ** cmd error: aborting 'source <abc-temp-dir>/abc.script'
ABC: Error: Cannot find buffer gate in the library.
ABC: Error: Abc_CommandAbc9Nf(): Mapping into LUTs has failed.
ERROR: Can't open ABC output file `/tmp/yosys-abc-XTTQQZ/output.blif'.
```

Note the warning from ABC: 
```ABC: Warnings: genlib library reader cannot detect the buffer gate.
ABC: Some parts of the supergate-based technology mapper may not work correctly.```

It appears that the liberty file you're using is too simplistic (I'm surprised it's enough for @cliffordwolf to build picorv32?!) -- after replacing the lib with something like this: https://vlsiarch.ecen.okstate.edu/flows/MOSIS_SCMOS/osu_soc_v2.5/cadence/lib/tsmc018/signalstorm/osu018_stdcells.lib I get a clean run.

Does this help?

cliffordwolf, 2019-03-02:
> I'm surprised it's enough for @cliffordwolf to build picorv32?!

It's not. It used to be but that was with an older version of ABC. I've now fixed that cell library accordingly in https://github.com/cliffordwolf/picorv32/commit/243a09fd8dc79e4435df145bc5f5dbaa95f210a5.

balanx, 2019-03-04:
thanks your reply !
It fixed this ERROR after I updated synth_gates.lib.

But, the synthesized result is NOT the simplest !! @cliffordwolf
I supplied an issue at 'https://github.com/YosysHQ/yosys/issues/765'.

closed 2019-03-04 by eddiehung