YosysHQ/yosys #825: read_library bus pin support

mvphilip, 2019-02-22:
Hello, I have a bus pin that read_liberty seems not handle:

  type( IO_bus_3_to_0 ) {
     base_type : array ;
     data_type : bit ;
     bit_width : 4;
     bit_from : 3 ;
     bit_to : 0 ;
     downto : true ;
  }

		bus(I1)  {
			bus_type : IO_bus_3_to_0 ;
			direction : input;
			pin (I1[3:0]) {

The error is on the last line:
ERROR: Syntax error in liberty file on line 28.

## Steps to reproduce the issue

[sram.lib.gz](https://github.com/YosysHQ/yosys/files/2895623/sram.lib.gz)

yosys
read_liberty -lib -ignore_miss_func -ignore_miss_data_latch -ignore_miss_dir -setattr blackbox sram.lib

## Expected behavior

The original library is valid and is accepted by commercial tools.

## Actual behavior

An error is reported on line 28:

Thank you,

-MP

mguthaus, 2019-02-23:
Adding a comment to follow developments. I've run into this error with OpenRAM support.

mvphilip, 2019-02-23:
Easy work around is to simply expand the pin members within the bus statement. It will still be recognized as a bus. It will make liberty much larger as arcs are now duplicated however. For example:

	bus(I1)  {
		bus_type : IO_bus_3_to_0 ;
		direction : input;
		pin (I1[3]) { .. }
                pin(I1[2]) { .. }
		pin (I1[1]) { .. }
                pin(I1[0]) { .. }
      }

Thanks,

-mp

eddiehung, 2019-12-07:
Looks like this was fixed by #897 and did not get closed.
Confirmed on 7dece795 to give this output:
```
 Yosys 0.9+932 (git sha1 a7e0cca4, clang 9.0.0-2 -fPIC -Os)

-- Running command `read_liberty -lib -ignore_miss_func -ignore_miss_data_latch -ignore_miss_dir -setattr blackbox sram.lib; dump' --

1. Executing Liberty frontend.
Imported 1 cell types from liberty file.

autoidx 1

attribute \blackbox 1
module \SRAM

  wire width 4 input 2 \I1

  wire input 1 \CE1
end
```

closed 2019-12-07 by eddiehung