YosysHQ/yosys #826: [regression]

daveshah1, 2019-02-24:


adumont, 2019-02-24:
Hi. My project passed building (synthesizing) last week just fine (https://travis-ci.org/adumont/hrm-cpu/builds/494525894), it was using yosys at commit e45f62b0c56717a23099425f078d1e56212aa632. 

Today, at Yosys commit  c521f4632f1c82b48a5538c832980668044e8fd9, my same code is failing to synthesize with error: 

`hrmcpu.v:146: ERROR: Identifier `\program0.PROGRAM' is implicitly declared and `default_nettype is set to none.`

Offending line in my code according to yosys is: https://github.com/adumont/hrm-cpu/blob/master/verilog/hrmcpu.v#L146:

`defparam program0.PROGRAM = PROGRAM;`

Changing to PROG #(.PROGRAM(PROGRAM)) program0 (...) seems to work as a workaround.

--

confirmed, synthesis goes well with yosys @ 974927ad (previous commit), then fails with yosys @ 23148ffa.

cliffordwolf, 2019-02-24:
Ooops. Fixed in 1816fe0.

closed 2019-02-24 by cliffordwolf

adumont, 2019-02-24:
Fix confirmed, working now

https://travis-ci.org/adumont/hrm-cpu/builds/497851251

Thanks a lot