YosysHQ/yosys #828: Incorrectly generated verilog

renau, 2019-02-26:
## Steps to reproduce the issue

When I read the verilog code for dce2, with

read_verilog dce2.v
proc
opt
write_verilog dce2_opt.v

I run check for dce2.v and dce2_opt.v and they are not equivalent. (both formality and yosys show that the code is not equivalent)

module dce2(input clk, input a, input b, output d);
reg [1:0] array1;
reg [1:0] array2;

reg c;
always @(*) begin
  c = a ^b;
end

always @(posedge clk) begin
  array1[c] <= array1[c]^a;
  array2[c] <= array2[c]^a;
  d <= array1[~c];
end
endmodule

renau, 2019-02-26:
I forgot to mention that this fails with top of the tree (and older yosys versions)
This simpler code also fails:

module dce3(input clk, input b, input a, output d);
  reg [1:0] array1;

  always @(posedge clk) begin
    d <= array1[a]^b;
    array1[a] <= a^b;
  end

endmodule

cliffordwolf, 2019-02-28:
Hmm.. I can't reproduce the problem:

```
clifford@euler:~/Work/yosys$ bash tests/tools/autotest.sh -p 'proc; opt' dce3.v 
Test: dce3 -> ok
clifford@euler:~/Work/yosys$ bash tests/tools/autotest.sh -p 'proc; opt' dce2.v 
Test: dce2 -> ok
```

This is after `s/output d/output reg d/`, but yosys does the same in both cases. It's just that without the `reg` yosys will produce a warning but icarus verilog will produce an error. (Afaict yosys produces exactly the equivalent output with and without the `reg`, so that doesn't seem to be the issue.)

Can you elaborate: What's the input sequence for which you think Yosys produces an output that isn't equivalent to the input design?

renau, 2019-03-01:
I have a file called dce3.v (previous example). I do this:

read_verilog -sv dce3.v ; proc ; opt -fast ; pmuxtree ; memory_dff ; memory_share ; memory_collect
write_verilog dce3b.v

If I run to check the equivalence between dce3b.v and dce3.v I get a failure. The generate code for dce3b.v (without comments)
```
module dce3(clk, b, a, d);
  wire [1:0] _00_;
  wire _01_;
  wire _02_;
  wire _03_;
  input a;
  reg [1:0] array1;
  input b;
  input clk;
  output d;
  reg d;
  always @(posedge clk)
      d <= _01_;
  always @(posedge clk)
      array1 <= _00_;
  assign _00_[1] = a ? (* src = "dce3.v:6" *) _03_ : array1[1];
  assign _00_[0] = a ? (* src = "dce3.v:6" *) array1[0] : _03_;
  wire [1:0] _11_ = array1;
  assign _02_ = _11_[a +: 1];
  assign _01_ = _02_ ^ (* src = "dce3.v:5" *) b;
  assign _03_ = a ^ (* src = "dce3.v:6" *) b;
endmodule
```

The source of the problem is the assign _02_
If I change it to this:
  assign _02_ = _11_[a];

Then, both yosys equivalence check and formality pass correctly

(to tell you the thruth, they look like the same verilog, but even yosys check with equiv does not say so)

cliffordwolf, 2019-03-02:
Well I can't reproduce what you are seeing because you don't tell how you do the formal equivalence check with yosys. What exactly is your expectation here?

But I can tell you that `_11_[a +: 1]` and `_11_[a]` should be exactly identical.

Maybe there's a bug or user error in the equivalence check. But I can't tell because you don't say what you are doing there..

However, I did create my own equivalence checking script and it proves equivalence between the synthesized dce3 from your last post with `_11_[a +: 1]` and `_11_[a]`: [dce3_eq.zip](https://github.com/YosysHQ/yosys/files/2922329/dce3_eq.zip)

So I'm closing the issue now. I've sunk enough time in this and even though I try very hard there is no bug that I could reproduce given the information you provided.

closed 2019-03-02 by cliffordwolf