YosysHQ/yosys #840: Make test Fails at svinterfaces

Kish0re7, 2019-03-01:
cd tests/svinterfaces && bash run-test.sh ""
Test: svinterface1 -> Segmentation fault (core dumped)
mv: cannot stat 'output.txt': No such file or directory
Segmentation fault (core dumped)
mv: cannot stat 'output.txt': No such file or directory
diff: reference_result.txt: No such file or directory
diff: dut_result.txt: No such file or directory
ERROR!
Test: svinterface_at_top -> Segmentation fault (core dumped)
mv: cannot stat 'output.txt': No such file or directory
Segmentation fault (core dumped)
mv: cannot stat 'output.txt': No such file or directory
diff: reference_result.txt: No such file or directory
diff: dut_result.txt: No such file or directory
ERROR!
Makefile:570: recipe for target 'test' failed
make: *** [test] Error 255

cliffordwolf, 2019-03-02:
Insufficient information to reproduce the issue. It works here (Ubuntu 18.04):

```
clifford@euler:~/Work/yosys$ cd tests/svinterfaces/
clifford@euler:~/Work/yosys/tests/svinterfaces$ bash run-test.sh 
Test: svinterface1 -> ok
Test: svinterface_at_top -> ok
```

And it also works in Travis CI.

closed 2019-03-02 by cliffordwolf

BharathS11, 2019-05-17:
Same error.
Fresh install of ubuntu 18.04 on a virtual machine. Building yosys from git. Fails on svinterfaces test.
I have the crash dump, if it can help.

Thanks

mmicko, 2019-05-17:
@BharathS11 Make sure you are using latest iverilog, packaged one is very old version, you take get sources for latest one from : https://github.com/steveicarus/iverilog

BharathS11, 2019-05-17:
@mmicko That was it. Thanks