YosysHQ/yosys #844: opt_clean removing all blackbox cells

Fatsie, 2019-03-02:
## Steps to reproduce the issue

I am using proprietary files so I can't include all files to reproduce the problem.
Attached is [SnowWhiteI.zip](https://github.com/YosysHQ/yosys/files/2922114/SnowWhiteI.zip) with the yosys script and the log file which should show the problem.
I am using TSMC libraries and am successfully able to do dfflibmap and abc with them. I need to specify abc script as the default crashes abc.
As one can see the opt_clean will

## Expected behavior

Cells that are from 'read_liberty -lib' being kept by opt_clean

## Actual behavior

cells removed

## yosys version

Yosys 0.8+209 (git sha1 fb84252, gcc 6.3.1 -fPIC -Os)
This is on top of 57f8bb4 with a few local.

cliffordwolf, 2019-03-02:
> Cells that are from 'read_liberty -lib' being kept by opt_clean

`read_liberty -lib` doesn't read any cells. That's what -lib does. `opt_clean` doesn't care if a cell instantiates a blackbox or a full module. `opt_clean` does not remove modules.

The .zip file you attached doesn't contain a design, just a log file and a yosys script. So it's impossible to reproduce the problem. And the logfile is 74k lines long and you don't even make an effort to point out where in the logfile you think the problem manifests.

This issue contains no actionable information. Therefore I'm closing it.

Please provide an MCVE and we can have a look.

closed 2019-03-02 by cliffordwolf

Fatsie, 2019-03-02:
Hope I may do a little discussion on a closed issue (keeping closed for the moment)
Sorry I was not clear enough. In the script you see I included a stat before and after opt_clean. The first stat is thus after abc.
Output of first stat starts on line 46359 and gives

     === SnowWhite_Top ===
    
       Number of wires:              50677
       Number of wire bits:          58445
       Number of public wires:        1817
       Number of public wire bits:    8221
       Number of memories:               0
       Number of memory bits:            0
       Number of processes:              0
       Number of cells:              26277

Second stat start on line 74451:

    === SnowWhite_Top ===
    
       Number of wires:                 91
       Number of wire bits:             91
       Number of public wires:          91
       Number of public wire bits:      91
       Number of memories:               0
       Number of memory bits:            0
       Number of processes:              0
       Number of cells:                  0

So all standard cells inserted by abc are removed by opt_clean.
Is this expected behaviour ?
What is rule for opt_clean to remove blackbox cells ?

Problem with providing MCVE is that I am using proprietary files I am not allowed to share.

cliffordwolf, 2019-03-03:
> Is this expected behaviour ?

It is the expected behavior if they for example don't contribute to an output signal. But that has nothing to do with them being blackbox cells.

> What is rule for opt_clean to remove blackbox cells ?

As I said before, there is no rule for opt_clean for blackbox cells. opt_clean simply does not distinguish between blackbox cells and other types of cells.

> Problem with providing MCVE is that I am using proprietary files I am not allowed to share.

It probably wouldn't be an MCVE if you send your whole design anyways, considering the M stands for minimal.

If you can't provide an example I can run then obviously there is nothing I can do.

Fatsie, 2019-03-03:
Thanks, the cells do contribute to the output of the block so I will try to make MCVE using FreePDK.

Fatsie, 2019-03-03:
Sorry, my bad. Top cell was generated by migen and had only input ports.

cliffordwolf, 2019-03-03:
:laughing: No worries.