YosysHQ/yosys #853: yosys-smtbmc: Contradictory "Cover" Results

hotwolf, 2019-03-04:
This is a repost of issue [#706](https://github.com/YosysHQ/yosys/issues/706)

I have a simple [design ](https://github.com/hotwolf/WbXbc/blob/master/rtl/verilog/WbXbc_address_decoder.v) with a[ test bench](https://github.com/hotwolf/WbXbc/blob/master/bench/verilog/ftb_WbXbc_address_decoder.sv), that contains two cover statements:

```
cover (wb_itr_mon_fsm_busy & $past(wb_itr_mon_fsm_busy));
cover (wb_itr_mon_fsm_busy & ~rst & req & ack);
```
The first one (trying to prove, that the bus monitor can assert the signal "wb_itr_mon_fsm_busy" for two consecutive cycles) is reported as unreachable.
The second cover statement produces a witness trace, showing signal "wb_itr_mon_fsm_busy" asserted for two consecutive cycles (pls. see [screenshot](https://hotwolf.github.io/WbXbc/images/issue706.png)). This contradicts the result of the first cover statement.

To run my test case, please execute the following steps:
```
git clone https://github.com/hotwolf/WbXbc.git
cd WbXbc
make cover.WbXbc_address_decoder
make debug
```

(Note: "make debug" requires the installation of Perl and the Verilog-Perl library. Alternatively, run "make debug.list" to see the location of the VCD file and run gtkwave manually)

ZipCPU, 2019-03-05:
I was about to post my analysis of #706 , but it seems you've reposted.

Here's your problem: 
1. The witness for your original cover() statement, the one without the $past, succeeds with no problems.  It succeeds, however, *on the last clock of the given depth*, and so it succeeds at the last chance it could succeed.
2. The SBY file then directs the formal solver to append an additional 8 steps to the witness.  During these 8 steps, the second cover is made to be true.  However, since it isn't within the original depth, this second cover statement does not pass.
3. If you increase the cover depth to 14, both statements pass.

I'm therefore going to recommend that this issue also be closed.  The tool is working as intended.

closed 2019-03-05 by cliffordwolf

hotwolf, 2019-03-05:
You are correct. I had overlooked that part of the trace had been created by the "append" statement. Thanks for helping me out with this, I would have never figured this out on my own.