YosysHQ/yosys #855: Keep attribute does not duplicate all cells

pmassolino, 2019-03-05:
I need to compute the following:
- nc = (a&b) ^ c;
- nd = (a&b) ^ d;

However, I would like to force the tool to not optimize resources and keep all the gates.
I added the keep attribute and then I noticed that when I use the vsclib013.lib ( not attached ) it does as expected and replaces nc/nd with an AND+XOR gate. However, when I am using a library that does not have this gate gscl45nm.lib ( not attached ) and it has to use NAND gate instead, it performs resource sharing of the a&b operation with a NAND gate and then duplicates the inverter.

[test_with_keep.zip](https://github.com/YosysHQ/yosys/files/2932487/test_with_keep.zip)

In the zip file there should be:

- nano_test.v which is the verilog source with the gates.
- nano_test1.ys which is the yosys script for vsclib013.lib
- nano_test2.ys which is the yosys script for gscl45nm.lib
- vsclib013.contr/gscl45nm.contr wich are the pin constraint files.

I tested with version 0.8 and the github master.

daveshah1, 2019-05-18:
The Yosys README says
> The keep attribute on cells and wires is used to mark objects that should never be removed by the optimizer. This is used for example for cells that have hidden connections that are not part of the netlist, such as IO pads. Setting the keep attribute on a module has the same effect as setting it on all instances of the module.

Although keep will tell Yosys not to optimise away things like cell instances altogether, it is not intended to prevent optimisations across modules boundaries. Yosys provides `(* keep_hierarchy *)` which will prevent an instance being flattened and therefore prevent any across-boundary optimisations. If you need a flat netlist at the end, you might need to remove this attribute from the instances using `attrmap -remove` before a final `flatten` after all synthesis and optimisation.

pmassolino, 2019-05-19:
Thanks for the response. I will definitely use `(* keep_hierarchy *)` attribute instead. I was confused when to use one or another.
Also, thanks for telling about the ` attrmap -remove` , because it might come in handy.

closed 2019-05-19 by pmassolino