YosysHQ/yosys #857: Duplicated reg after read_verilog, where task is used inside

nobodybutyou1, 2019-03-07:
## Steps to reproduce the issue
a simple example is like
name: tc.v
module tc(data_in,data_out,clk); 
output reg  data_out; 
input  data_in; 
input clk; 

always @(posedge clk) 
begin 
    myTask(data_out,data_in);
end 
task myTask; 
output out; 
input in; 
    out = in; 
endtask 
endmodule

## Expected behavior
This should be 1 FF generated. 
*Please describe the behavior you would have expected from the tool.*

## Actual behavior
There are 2 FFs generated.
How to reproduce:
Within yosys type in the following commands:
###############
read_verilog tc.v
write_verilog result.v
###############
You'll see two REGs. I believe something goes wrong while parsing task in Verilog.
example:
 always @* begin
    _2_ = _4_;
    _0_ = data_in;
    _1_ = data_in;
  end
  always @(posedge clk) begin
      data_out <= _2_;
      _4_ <= _1_;
      _3_ <= 1'hx;
  end
endmodule

_2_ and _4_ are two REGs I got. At the meantime, commercial tool, such as Vivado generates 1 FF for this Verilog code.

(You can also go through the whole synthesis process and the see 2 FFs reported at the end of synthesis...)

cliffordwolf, 2019-03-08:
Wow. That's odd.
Thanks for the bug report. Looking into it now.

closed 2019-03-08 by cliffordwolf

nobodybutyou1, 2019-03-11:
It looks like an AST tree issue. Thanks for the quick fix on it!

commit a330c683 (2019-03-08): Fix handling of task output ports in clocked always blocks, fixes #857