YosysHQ/yosys #865: Got a different result by adding write_verilog right after read_verilog

nobodybutyou1, 2019-03-11:
## Steps to reproduce the issue
Sample code is as follows.

module tc(clk,A,B,C,D,E,F);
input clk;
input[3:0] A,B,C,E;
output reg[7:0] D,F;
always @(posedge clk)
		begin
			  D = A + B + C;
			F =  A + B + E;
		end
endmodule

The following two command sets are supposed to generate the same results.
tcl set 1 (reported 14 FDRE)     filename: run1.tcl
yosys read_verilog tc.v
yosys proc
yosys opt
yosys fsm
yosys opt
yosys memory
yosys opt
yosys synth_xilinx -top tc

tcl set 2 (reported 12 FDRE)     filename: run2.tcl
yosys read_verilog tc.v
yosys write_verilog temp.v
yosys proc
yosys opt
yosys fsm
yosys opt
yosys memory
yosys opt
yosys synth_xilinx -top tc

how to run the tcl files: "yosys -c run1.tcl" and "yosys -c run2.tcl"

## Expected behavior

After the synthesis, two command sets are supposed to generate the same results. 
(12 FDREs is the result we expect, cause D[7:6] and F[7:6] are always 0s.)

## Actual behavior

tcl set1 (run1.tcl) reported 14 FDREs, 

=== tc ===

   Number of wires:                 41
   Number of wire bits:             99
   Number of public wires:           7
   Number of public wire bits:      33
   Number of memories:               0
   Number of memory bits:            0
   Number of processes:              0
   Number of cells:                 73
     FDRE                           14
     LUT2                           24
     MUXCY                          16
     XORCY                          19

whereas tcl set2 (run2.tcl) reported 12 FDREs.

   Number of wires:                 41
   Number of wire bits:             95
   Number of public wires:           7
   Number of public wire bits:      33
   Number of memories:               0
   Number of memory bits:            0
   Number of processes:              0
   Number of cells:                 67
     FDRE                           12
     LUT2                           24
     MUXCY                          14
     XORCY                          17

cliffordwolf, 2019-03-11:
I think #866 will fix this example.

Some background information: First, the difference in result is just random noise resulting from different sort orders for some nets. The internal netlist is exactly the same before and after `write_verilog`. The only difference is that `write_verilog` will allocate and deallocate some IDs, slightly changing the numerical values allocated to other IDs down the line, which then changes sort order.

However, it's not a "real" determinism issue because you are comparing two runs with different scripts. We only consider something a problem with determinism if the same script, run on different platforms, or on yosys binaries built with different compilers, or simply because of address space randomization, exhibit different behavior. This is not the case here.

In #866 we change some things to slightly increase the determinism in this case, and we also add `YOSYS_SORT_ID_FREE_LIST` to `kernel/rtlil.h`. You can change this setting from `#undef` to `#define` for further increased determinism at a small extra cost in runtime performance.

closed 2019-03-11 by cliffordwolf

nobodybutyou1, 2019-03-12:
Thanks, I see your point. My concern is that the changes you made lead to consistency but affect optimization. In this example, the highest two bits of either D or F are always 0, which should be optimized away after synthesis. The remaining number of FFs is 8(width of D=8)+8(width of F=8)-2(highest two bits of D are 0s)-2(highest two bits of F are 0s) = 12, instead of 14 (Yosys result after your changes). I tried other two commercial tools, both reported 12 FFs in this case.

nobodybutyou1, 2019-03-14:
I got it fixed by doing multiple rounds of "rewrite"...