YosysHQ/yosys #867: ROM memory inference and initialization

mkurc-ant, 2019-03-12:
Starting from merge of #843 it is not possible to force Yosys to infer a ROM memory from a verilog array initialized using `initial` statement(s).

From my understanding the goal of #843 was optimization so small ROMs are always implemented using LUTs. That makes sense. However in my case I want to place a CPU program inside an initialized BRAM and I want the BRAM to be inferred rather instantiated explicitly.

Please find a test case attached: [bram-init.zip](https://github.com/YosysHQ/yosys/files/2957512/bram-init.zip) This is a ROM memory with firmware intended to be used by the PicoSoC.

I suppose that there should be kind of a "threshold" so the Yosys can decide: If a memory is ROM and is smaller than X bytes then use LUTs, otherwise make it a ROM memory.

mkurc-ant, 2019-03-12:
[bram-init.zip](https://github.com/YosysHQ/yosys/files/2957565/bram-init.zip)

cliffordwolf, 2019-03-12:
Ooops. Thanks for reporting this issue. Should be fixed in #868. Please test and report your findings.

mkurc-ant, 2019-03-13:
@cliffordwolf Checked with #868 and the BRAM inference is back to normal. Thanks.

When synthesizing project from the `bram-init.zip` I posted, the result is:
>    Number of wires:                 14
   Number of wire bits:            185
   Number of public wires:           9
   Number of public wire bits:     111
   Number of memories:               0
   Number of memory bits:            0
   Number of processes:              0
   Number of cells:                  4
     FDCE                            1
     LUT1                            1
     LUT2                            1
     RAMB36E1                        1

which is what was expected.

commit ab5b50ae (2019-03-13): Only set MEM2REG_FL_CONST_LHS/MEM2REG_FL_VAR_LHS for non-init writes, fixes #867
closed 2019-03-13 by cliffordwolf