YosysHQ/yosys #870: Yosys crashes on reading exported verilog design containing concatenation.

aJunk, 2019-03-13:
Having run `opt -full` after `flatten` yosys crashes on importing just exported verilog file.

The error message is:

> 1. Executing Verilog-2005 frontend.
Parsing Verilog input from `multimux_out_2.v' to AST representation.
ERROR: Parser error in line multimux_out_2.v:34: syntax error, unexpected '['

## Steps to reproduce the issue

```
read_verilog multimux.v
hierarchy -auto-top -check;
proc; clean;
memory;
opt -full;
flatten;

write_verilog multimux_out_1.v
opt -full;
write_verilog multimux_out_2.v

delete;
read_verilog multimux_out_2.v
```

## Expected behavior

Yosys does not crash on multimux_out_1.v.

## Actual behavior
Yosys crashes on multimux_out_2.v

The difference between the generated verilog is that \m3.q is now concatenated and accessed through indexing in the same line:

`assign \m3.q  = { \m2.q , \m1.q  }[select[0] +: 1];`

Yosys does still generate  `assign q_tmp = { \m2.q , \m1.q  };`   
but does not use it.

If I insert q_tmp into the assignment of \m3.q it works again.

The input file and the two output files I got are attached.

[multimux.zip](https://github.com/YosysHQ/yosys/files/2962973/multimux.zip)

aJunk, 2019-03-13:
I also tried to process the exported verilog using [verilator](https://www.veripool.org/wiki/verilator) which yielded a similar error.

udif, 2019-03-13:
As much as we would like, 
```
assign \m3.q = { \m2.q , \m1.q }[select[0] +: 1];
```
Is not a legal Verilog or SystemVerilog syntax.
packed/unpacked array indexing only works for registers/wires, not for arbitrary expressions.

aJunk, 2019-03-13:
So you are saying that `{ \m2.q , \m1.q }` is not a wire which is why it can not be indexed?
That explains why it works when it is first assigned to the q_tmp wire.

But that still leaves to discover why it is exported that way.

aJunk, 2019-03-14:
I narrowed it down to the handling of $shiftx in the verilog backend where the unsupported expression is generated.

The issue has already been fixed in [this commit ](https://github.com/YosysHQ/yosys/commit/6c5049f016538e887476bb66d3f653155fa354ff#diff-b2d003f2d72289e7ee3972c7850735a4) by cliffordwolf in January.

The project we are working on modifies and compiles yosys itself and was still using a slightly older version because it does not pull in the changes automatically.

Thank you for your time an effort.

aJunk, 2019-03-14:
Everything works as intended now. The backend correctly generates an intermediate signal.

Thank you very much for your time and effort!

closed 2019-03-14 by cliffordwolf