YosysHQ/yosys #879: Verific Frontend Error

OrDavidi, 2019-03-16:
Hi, I got this Error: 

ERROR: Can't import Verific OPER_SHIFT_RIGHT instance shift_right_3: carry_in is neither 0 nor msb of left input

Reproduce : 
Run any verific pass on this module : 
[The weird thing is that if i remove the 2nd instance (gck) it passed]

module yosys_verific_shr_carry
  (
   output [4-1:0] b_s, 
   input 	  cc
   );
   wire [4-1:0] 	   b_s_pre;
   assign b_s             = b_s_pre >> cc ;
aaaa aaaaa
  (
   .s(b_s_pre)
   );   
  gck gck 
    (
     .sm	(1'b0)
    );
endmodule 

Thanks

eddiehung, 2019-03-16:
I'm not seeing what you're seeing, on Yosys git with Feb19 Verific:
```
$~/yosys/yosys -p 'verific -sv top.v; verific -import -all'

<<SNIP>>

-- Running command `verific -sv top.v; verific -import -all' --

1. Executing VERIFIC (loading SystemVerilog and VHDL designs using Verific).
Built with Verific Feb19_SW_Release, released at Fri Mar  1 15:28:43 2019.
VERIFIC-COMMENT [VERI-1482] Analyzing Verilog file 'top.v'

2. Executing VERIFIC (loading SystemVerilog and VHDL designs using Verific).
Built with Verific Feb19_SW_Release, released at Fri Mar  1 15:28:43 2019.
Running hier_tree::ElaborateAll().
VERIFIC-INFO [VERI-1018] top.v:1: compiling module 'yosys_verific_shr_carry'
VERIFIC-ERROR [VERI-1063] top.v:11: instantiating unknown module 'aaaa'
VERIFIC-INFO [VERI-1073] top.v:1: module 'yosys_verific_shr_carry' remains a black box, due to errors in its contents
ERROR: top.v:11: instantiating unknown module 'aaaa'
```

Can you share your exact script, and the version of Yosys you are using?

e.g. ` Yosys 0.8+254 (git sha1 514dda5, clang 3.8.0-2ubuntu4 -fPIC -Os)`
and `Built with Verific Feb19_SW_Release, released at Fri Mar  1 15:28:43 2019`

OrDavidi, 2019-03-16:
yosys -p "verific -sv yosys_verific_shr_carry.v" -p "verific -import yosys_verific_shr_carry"

re the version , how to extract this info ? 

Thanks

eddiehung, 2019-03-16:
That command also works for me:
```
$ yosys -p "verific -sv yosys_verific_shr_carry.v" -p "verific -import yosys_verific_shr_carry"

 /----------------------------------------------------------------------------\
 |                                                                            |
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |                                                                            |
 |  Copyright (C) 2012 - 2018  Clifford Wolf <clifford@clifford.at>           |
 |                                                                            |
 |  Permission to use, copy, modify, and/or distribute this software for any  |
 |  purpose with or without fee is hereby granted, provided that the above    |
 |  copyright notice and this permission notice appear in all copies.         |
 |                                                                            |
 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
 |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
 |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
 |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
 |                                                                            |
 \----------------------------------------------------------------------------/

 Yosys 0.8+254 (git sha1 514dda5, clang 3.8.0-2ubuntu4 -fPIC -Os)

-- Running command `verific -sv yosys_verific_shr_carry.v' --

1. Executing VERIFIC (loading SystemVerilog and VHDL designs using Verific).
Built with Verific Feb19_SW_Release, released at Fri Mar  1 15:28:43 2019.

-- Running command `verific -import yosys_verific_shr_carry' --

2. Executing VERIFIC (loading SystemVerilog and VHDL designs using Verific).
Built with Verific Feb19_SW_Release, released at Fri Mar  1 15:28:43 2019.
Adding Verilog module 'yosys_verific_shr_carry' to elaboration queue.
Running hier_tree::Elaborate().
VERIFIC-INFO [VERI-1018] yosys_verific_shr_carry.v:1: compiling module 'yosys_verific_shr_carry'
VERIFIC-ERROR [VERI-1063] yosys_verific_shr_carry.v:11: instantiating unknown module 'aaaa'
VERIFIC-INFO [VERI-1073] yosys_verific_shr_carry.v:1: module 'yosys_verific_shr_carry' remains a black box, due to errors in its contents
ERROR: yosys_verific_shr_carry.v:11: instantiating unknown module 'aaaa'
```

That information can be extracted from the first line after the copyright message, and after the first `verific` command (i.e. `Built with Verific Feb19_SW_Release, released at Fri Mar  1 15:28:43 2019.`)

OrDavidi, 2019-03-16:
Oh i see, 
I have earlier version of verific - i will update.

Thanks!

closed 2019-03-16 by OrDavidi

eddiehung, 2019-03-16:
I wouldn't be so hasty to close just yet! 
For the record, what version are you using?