YosysHQ/yosys #894: Relaxing of newline results in strict ';' need

Fatsie, 2019-03-23:
In 04cd179 newline handling was relaxed but it results in more strict need of ';' usage.
I have a .lib file from TSMC where not each line ends in ; and this now fails in yosys, e.g.

    cell (XXX) {
      ...
      area : 87.5
      cell_footprint : "XXX"
      pin (A1) {
      ...
    }

Reverting 04cd179 makes the TSMC .lib work again.

trcwm, 2019-03-24:
TSMC is using a incorrect .lib syntax 😃 Will fix asap.

Fatsie, 2019-03-25:
BTW, culprit is likely Library Compiler of Synopsys itself. The std. cell lib also includes 'compiled' liberty files with .db extension. This means they went through the Synopsys Library Compiler without problem.

Fatsie, 2019-03-25:
I can confirm #897 works with the TSMC liberty file.

closed 2019-03-27 by cliffordwolf