YosysHQ/yosys #898: Missing parameter assignment results in Warning: No clocks found in design

gojimmypi, 2019-03-25:
## Steps to reproduce the issue

Please see the details of https://github.com/YosysHQ/nextpnr/issues/253 - this is probably more of a feature request than an actual bug.  I've created this issue in yosys per a [comment](https://github.com/YosysHQ/nextpnr/issues/253#issuecomment-475432514) by @koriakin 

> Note that nextpnr doesn't see your original design -- by the time yosys is done with your design, there really is no register left on the clock, and this is what nextpnr sees. It also cannot see that you don't assign to oled, because yosys simplifies that to a constant driver using the initial value (which is a perfectly reasonable design as far as nextpnr is concerned). If you would like a better warning about that, it needs to be in yosys.

## Expected behavior

I'm hoping for a more explicit message than simply `Warning: No clocks found in design` when, as @daveshah1 [noted](https://github.com/YosysHQ/nextpnr/issues/253#issuecomment-475146011):

> This warning is generated when there are no interior paths (ie paths between register output and input) for any clock in the design, as [note here].(https://github.com/YosysHQ/nextpnr/issues/253#issuecomment-475343715) 

## Actual behavior

All (?) issues for "no interior path" apparently result in the same `Warning: No clocks found in design` message. It would be really useful to actually _name the problematic element_.

Admittedly, I'm an FPGA newbie, and this was probably a bonehead mistake by any measure. Still, I spent a tremendous amount of time googling clock issues, when in fact the actual problem was unrelated to the clock. 

If things like this are more easily self resolved with detailed error messages, I think that would help tremendously to expand the use of FPGA in the technical community.  :)

cliffordwolf, 2019-03-27:
This issue is not actionable. The expected/actual behavior section doesn't describe what change OP want's to perform, and only refers to a nextpnr warning message, not to any desired behavior change in Yosys.

Testing with the code from YosysHQ/nextpnr#253, with the assignment to `o_led[5:0]` commented out, I receive the following warning messages from Yosys:

```
$ yosys -ql blinky.log blinky.ys
Warning: Wire top.\led [5] is used but has no driver.
Warning: Wire top.\led [4] is used but has no driver.
Warning: Wire top.\led [3] is used but has no driver.
Warning: Wire top.\led [2] is used but has no driver.
Warning: Wire top.\led [1] is used but has no driver.
Warning: Wire top.\led [0] is used but has no driver.
```

Furthermore, the logfile describes exactly what is happened to the `ctr` register:

```
2.4.7. Executing PROC_DFF pass (convert process syncs to FFs).
...
Creating register for signal `\top.\ctr' using process `\top.$proc$blinky.v:26$1'.
  created $dff cell `$procdff$103' with positive edge clock.
...

2.8.3. Executing OPT_CLEAN pass (remove unused cells and wires).
Finding unused cells or wires in module \top..
...
  removing unused `$dff' cell `$procdff$103'.
```

So I'm closing this issue as "not actionable" or "not a bug".

closed 2019-03-27 by cliffordwolf

gojimmypi, 2019-03-27:
@cliffordwolf thanks for your time in reviewing and explaining this. 

I agree with your assessment as "not actionable", as I confirmed `yosys` indeed outputs the warnings in 2.8.4 for me:

```
2.8.4. Executing CHECK pass (checking for obvious problems).
checking module top..
Warning: Wire top.\led [5] is used but has no driver.
Warning: Wire top.\led [4] is used but has no driver.
Warning: Wire top.\led [3] is used but has no driver.
Warning: Wire top.\led [2] is used but has no driver.
Warning: Wire top.\led [1] is used but has no driver.
Warning: Wire top.\led [0] is used but has no driver.
found and reported 6 problems.
```

I still think the corresponding `Warning: No clocks found in design` is a bit undesired in `nextpnr`, but now I understand better how `nextpnr` also does not "see" the warnings from `yosys`.

I wonder if somehow the warnings from `yosys` can be propagated to `nextpnr` for better insight? Or perhaps `nextpnr` warnings can simply say "_see output from yosys_".