YosysHQ/yosys #899: Optimization too aggressive?

mickaeltorres, 2019-03-25:
## Steps to reproduce the issue

Synthesizing this:

```
module TOP(input CLK, output reg LED);

reg [31:0] r[4];
reg [1:0]  rd;
reg [1:0]  step;

initial
begin
  r[0] = 0;
  r[1] = 0;
  r[2] = 0;
  r[3] = 0;
  rd = 0;
  step = 0;
end

always @(posedge CLK)
begin
  case (step)
    0: begin
      step <= 1;
      rd <= 0;
    end
    1: begin
      if (rd != 0)
        r[rd] <= 1;
      step <= 2;
    end
    2: begin
      if (r[1] == 1)
        LED <= 1;
      rd <= 1;
      step <= 1;
    end
  endcase
end

endmodule
```

with Yosys 0.8+ 299 (git sha1 ccfa2fe0, clang 6.0.0 -fPIC -Os):

```
yosys -p "read_verilog -noautowire top.v" -p 'synth_ice40 -top top -blif top.blif'
```

results in an empty design.

## Expected behavior

Design is not empty.
LED goes to 1, the second time the FSM is in state 2.

## Actual behavior

Design is empty.

daveshah1, 2019-03-25:
The problem is `r[4]`. Although I think some variants of SystemVerilog support C-style arrays, with traditional Verilog bounds this creates a single element with index 4.

What you want to use is `r[0:3]`.

Note that Yosys does produce a warning on the out-of-bounds access, running Yosys with -q can make it easier to see warnings.

mickaeltorres, 2019-03-26:
Thanks Dave, I guess I'm too used to SystemVerilog :D

closed 2019-03-26 by mickaeltorres