YosysHQ/yosys #900: Error with MSB bit in EDIF writer

Gl237man, 2019-03-27:
There's error during EDIF export.
If top-module has multi-bit input/output, EDIF writer during export always reverts bit order.
If bits of output port have different functions, then resulting function mapping will be incorrect.

Input verilog file:
```
(* blackbox *) 
module INV(A, Q);
input A;
output Q;
endmodule

(* blackbox *) 
module BUF(A, Q);
input A;
output Q;
endmodule

module testmodule (A, Q);
input [1:0] A;
output [1:0] Q;
INV inst0 (.A(A[0]), .Q(Q[0]));
BUF inst1 (.A(A[1]), .Q(Q[1]));
endmodule
```

Yosys script:

```
read_verilog testmodule.v
write_edif -nogndvcc dump.edf
write_verilog -noattr dump.v
```

Resulting Verilog file:

```
/* Generated by Yosys 0.8 (git sha1 5706e90, i686-w64-mingw32.static-g++ 5.4.0 -Os) */

module testmodule(A, Q);
  input [1:0] A;
  output [1:0] Q;
  INV inst0 (
    .A(A[0]),
    .Q(Q[0])
  );
  BUF inst1 (
    .A(A[1]),
    .Q(Q[1])
  );
endmodule
```

Resulting EDIF file:

```
    (cell testmodule
      (cellType GENERIC)
      (view VIEW_NETLIST
        (viewType NETLIST)
        (interface
          (port (array Q 2) (direction OUTPUT))
          (port (array A 2) (direction INPUT))
        )
        (contents
          (instance inst1
            (viewRef VIEW_NETLIST (cellRef BUF (libraryRef LIB))))
          (instance inst0
            (viewRef VIEW_NETLIST (cellRef INV (libraryRef LIB))))
          (net (rename id00001 "A[0]") (joined
            (portRef (member A 1))
            (portRef A (instanceRef inst0))
          ))
          (net (rename id00002 "A[1]") (joined
            (portRef (member A 0))
            (portRef A (instanceRef inst1))
          ))
          (net (rename id00003 "Q[1]") (joined
            (portRef (member Q 0))
            (portRef Q (instanceRef inst1))
          ))
          (net (rename id00004 "Q[0]") (joined
            (portRef (member Q 1))
            (portRef Q (instanceRef inst0))
          ))
        )
      )
    )
```

During Verilog export the program kept INV function on output Q[0].
But during EDIF export bits order changed, so INV functions is on portRef (member Q 1)).
If some external user desides to connect to the EDIF module depending on it's RTL functionality,
then his device will react unexpectedly.

But if invert bit order in input Verilog file, then in resulting EDIF order will be changed twice.

```
module testmodule (A, Q);
input [0:1] A;
output [0:1] Q;
INV inst0 (.A(A[0]), .Q(Q[0]));
BUF inst1 (.A(A[1]), .Q(Q[1]));
endmodule
```

Result:

```
    (cell testmodule
      (cellType GENERIC)
      (view VIEW_NETLIST
        (viewType NETLIST)
        (interface
          (port (array Q 2) (direction OUTPUT))
          (port (array A 2) (direction INPUT))
        )
        (contents
          (instance inst1
            (viewRef VIEW_NETLIST (cellRef BUF (libraryRef LIB))))
          (instance inst0
            (viewRef VIEW_NETLIST (cellRef INV (libraryRef LIB))))
          (net (rename id00001 "A[0]") (joined
            (portRef (member A 1))
            (portRef A (instanceRef inst1))
          ))
          (net (rename id00002 "A[1]") (joined
            (portRef (member A 0))
            (portRef A (instanceRef inst0))
          ))
          (net (rename id00003 "Q[1]") (joined
            (portRef (member Q 0))
            (portRef Q (instanceRef inst0))
          ))
          (net (rename id00004 "Q[0]") (joined
            (portRef (member Q 1))
            (portRef Q (instanceRef inst1))
          ))
        )
      )
    )
```

In this case output port Q[0] uses INV as expected.

Gl237man, 2019-03-27:
I suppose this problem is related
[https://github.com/YosysHQ/yosys/issues/568](url)

cliffordwolf, 2019-03-27:
See https://github.com/YosysHQ/yosys/pull/830#issuecomment-468809795 and http://svn.clifford.at/handicraft/2019/vivadoedif: The current behavior is to always use array index 0 for the MSB array element, and to use the name Verilog uses for the bit in the rename clause. This mirrors the exact behavior of Xilinx Vivado.

We can add support for other "EDIF flavors", but for that we would need to know what other tool you want us to emulate and what the exact behavior of that tool is.

closed 2019-03-27 by cliffordwolf