YosysHQ/yosys #902: Support for positional parameter argument without brackets

daveshah1, 2019-03-27:
Example code (minimised from some OpenSPARC code):
```
module add #(
	parameter N = 3
) (
	input [N-1:0] a, b,
	output [N-1:0] q
);
	assign q = a + b;
endmodule

module top(
	input [7:0] a, b,
	output [7:0] q
);
	add #8 add_i (.a(a), .b(b), .q(q));
endmodule
```

In Yosys this gives:
```
/home/david/ecp5/pp/pp.v:14: ERROR: syntax error, unexpected TOK_CONSTVAL, expecting '('
```

But other tools (LSE, Icarus & Verilator) seem to accept this.

udif, 2019-03-27:
Here is what the IEEE1800-2017 BNF says:

```
A.4 Instantiations
A.4.1 Instantiation
A.4.1.1 Module instantiation
module_instantiation ::=
module_identifier [ parameter_value_assignment ] hierarchical_instance { , hierarchical_instance } ;
parameter_value_assignment ::= # ( [ list_of_parameter_assignments ] )
```
The #(...) characters are actually tokens, not part of the BNF syntax. (The original document has them marked in RED, like all tokens).

Therefore, if your tool accepts this, it is allowing an incorrect syntax.

daveshah1, 2019-04-01:
Thanks for checking! It's interesting how widely accepted this syntax is, not sure if it is some kind of "de facto" thing or just a common parser bug...

closed 2019-04-01 by daveshah1