YosysHQ/yosys #904: missing cells using task in Verilog

nobodybutyou1, 2019-03-28:
## Steps to reproduce the issue
The Verilog is as follows.

module tc (A, B, CIN, S, COUT);
input [3:0] A, B;
input CIN;
output [3:0] S;
output COUT;
reg [3:0] S;
reg COUT;
reg [1:0] S0, S1, S2, S3;

task ADD;
input A, B, CIN;
output [1:0] C;
reg [1:0] C;
reg S, COUT;
  begin
    S = A ^ B ^ CIN;
    COUT = (A&B) | (A&CIN) | (B&CIN);
    C = {COUT, S};
  end
endtask

always @(A or B or CIN)
begin
  ADD (A[0], B[0], CIN, S0);
  ADD (A[1], B[1], S0[1], S1);
  ADD (A[2], B[2], S1[1], S2);
  ADD (A[3], B[3], S2[1], S3);
  S = {S3[0], S2[0], S1[0], S0[0]};
  COUT = S3[1];
end
endmodule

The yosys command to run the synthesis flow is as follows
read_verilog tc.v
proc
opt
fsm
opt
memory
opt
synth_xilinx -top tc

## Expected behavior

report cells such as LUT. 

## Actual behavior
report 0 cells.
This is reported by the latest yosys from GitHub.
=== tc ===

   Number of wires:                  5
   Number of wire bits:             14
   Number of public wires:           5
   Number of public wire bits:      14
   Number of memories:               0
   Number of memory bits:            0
   Number of processes:              0
   Number of cells:                  0

daveshah1, 2019-04-01:
Unable to reproduce this here. I get:
```
8.23. Printing statistics.

=== tc ===

   Number of wires:                 13
   Number of wire bits:             26
   Number of public wires:           9
   Number of public wire bits:      22
   Number of memories:               0
   Number of memory bits:            0
   Number of processes:              0
   Number of cells:                  9
     LUT2                            1
     LUT3                            3
     LUT5                            1
     LUT6                            3
     MUXF7                           1
```

Perhaps you could post the full log output?

cliffordwolf, 2019-04-22:
Looks like OP has abandoned the issue. Like @daveshah1 I can not reproduce this. So I'm closing it now.

PS: Running `proc; opt; fsm; opt; memory; opt` before running `synth_xilinx` doesn't make any sense, considering `synth_xilinx` will also run those passes. Not sure why OP has decided to use such a weird script. But either way, using OPs script, or using just `synth_xilinx` directly, I get the same results @daveshah1 posted.

closed 2019-04-22 by cliffordwolf