YosysHQ/yosys #906: Fix for reading bigger verilog files.

Kmanfi, 2019-03-29:
## Steps to reproduce the issue
Read some "big" verilog file.

## Expected behavior
Read.

## Actual behavior
Exit. Memory exhausted.

## Fix
```
#CXXFLAGS := $(CXXFLAGS) -Wall -Wextra -ggdb -I. -I"$(YOSYS_SRC)" -MD -D_YOSYS_ -fPIC -I$(PREFIX)/include
CXXFLAGS := $(CXXFLAGS) -Wall -DYYMAXDEPTH=100000 -Wextra -ggdb -I. -I"$(YOSYS_SRC)" -MD -D_YOSYS_ -fPIC -I$(PREFIX)/include
```

cliffordwolf, 2019-03-29:
Easy enough. ;) Please confirm that #907 fixes the issue.

Kmanfi, 2019-03-29:
Confirmed.

Kmanfi, 2019-05-18:
### Please increase YYMAXDEPTH by 100.
My design(about 1M gate design) hit this limit again and I increased -DYYMAXDEPTH by 100x to 10000000 and now it reads Verilog file in. _Confirmed fix._
```
frontends/verilog/verilog_parser.tab.o: CXXFLAGS += -DYYMAXDEPTH=100000
frontends/verilog/verilog_parser.tab.o: CXXFLAGS += -DYYMAXDEPTH=10000000
```
Would be nice to have a flag to enable debug prints for verilog parser when in debug mode.
Found this -FRONTEND_VERILOG_YYDEBUG=1

cliffordwolf, 2019-05-18:
Fixed in #1017

commit 584d2030 (2019-03-29): Build Verilog parser with -DYYMAXDEPTH=100000, fixes #906
closed 2019-05-18 by cliffordwolf