YosysHQ/yosys #908: MUX8 or MUX16 cells doesn't present in the netlist when use 'abc -mux8' or 'abc -mux16' command

SergeyDegtyar, 2019-04-01:
## Steps to reproduce the issue

When I use abc with -mux8 or -mux16 option then I don't have MUX8/MUX16 cells in the netlist:

yosys> abc -mux8

18. Executing ABC pass (technology mapping using ABC).

18.1. Extracting gate netlist of module `\top' to `<abc-temp-dir>/input.blif'..
Extracted 255 gates and 519 wires to a netlist network with 264 inputs and 1 outputs.

18.1.1. Executing ABC.
Running ABC command: <yosys-exe-dir>/yosys-abc -s -f <abc-temp-dir>/abc.script 2>&1
ABC: ABC command line: "source <abc-temp-dir>/abc.script".
ABC:
ABC: + read_blif <abc-temp-dir>/input.blif
ABC: + read_library <abc-temp-dir>/stdcells.genlib
ABC: Entered genlib library with 18 gates from file "<abc-temp-dir>/stdcells.genlib".
ABC: + strash
ABC: + ifraig
ABC: + scorr
ABC: Warning: The network is combinational (run "fraig" or "fraig_sweep").
ABC: + dc2
ABC: + dretime
ABC: + strash
ABC: + &get -n
ABC: + &dch -f
ABC: + &nf
ABC: + &put
ABC: + write_blif <abc-temp-dir>/output.blif

18.1.2. Re-integrating ABC results.
ABC RESULTS:               AND cells:       10
ABC RESULTS:            ANDNOT cells:       42
ABC RESULTS:              AOI3 cells:       25
ABC RESULTS:              AOI4 cells:        2
**ABC RESULTS:               MUX cells:      142**
ABC RESULTS:              NAND cells:       19
ABC RESULTS:               NOR cells:        3
ABC RESULTS:               NOT cells:        7
ABC RESULTS:              OAI3 cells:       76
ABC RESULTS:              OAI4 cells:       12
ABC RESULTS:                OR cells:       10
ABC RESULTS:             ORNOT cells:       11
ABC RESULTS:        internal signals:      254
ABC RESULTS:           input signals:      264
ABC RESULTS:          output signals:        1
Removing temp directory.

But when I use abc -mux4 for this case I have MUX4 cells:

yosys> abc -mux4

20. Executing ABC pass (technology mapping using ABC).

20.1. Extracting gate netlist of module `\top' to `<abc-temp-dir>/input.blif'..
Extracted 359 gates and 623 wires to a netlist network with 264 inputs and 1 outputs.

20.1.1. Executing ABC.
Running ABC command: <yosys-exe-dir>/yosys-abc -s -f <abc-temp-dir>/abc.script 2>&1
ABC: ABC command line: "source <abc-temp-dir>/abc.script".
ABC:
ABC: + read_blif <abc-temp-dir>/input.blif
ABC: + read_library <abc-temp-dir>/stdcells.genlib
ABC: Entered genlib library with 18 gates from file "<abc-temp-dir>/stdcells.genlib".
ABC: + strash
ABC: + ifraig
ABC: + scorr
ABC: Warning: The network is combinational (run "fraig" or "fraig_sweep").
ABC: + dc2
ABC: + dretime
ABC: + strash
ABC: + &get -n
ABC: + &dch -f
ABC: + &nf
ABC: + &put
ABC: + write_blif <abc-temp-dir>/output.blif

20.1.2. Re-integrating ABC results.
ABC RESULTS:               AND cells:       10
ABC RESULTS:            ANDNOT cells:       10
ABC RESULTS:              AOI3 cells:       19
ABC RESULTS:              AOI4 cells:        2
**ABC RESULTS:               MUX cells:       26
ABC RESULTS:              MUX4 cells:       56**
ABC RESULTS:              NAND cells:       14
ABC RESULTS:               NOR cells:        5
ABC RESULTS:               NOT cells:       10
ABC RESULTS:              OAI3 cells:       35
ABC RESULTS:              OAI4 cells:        6
ABC RESULTS:                OR cells:        9
ABC RESULTS:             ORNOT cells:        3
ABC RESULTS:        internal signals:      358
ABC RESULTS:           input signals:      264
ABC RESULTS:          output signals:        1
Removing temp directory.

Is it bug or I do something wrong?

###**Test case:**

module top (
input [7:0] S,
input [255:0] D,
output M256
);

assign M256 = D[S];

endmodule

### **Script:**

read_verilog ../top.v
synth -top top
abc -mux8

cliffordwolf, 2019-04-30:
> Is it bug or I do something wrong?

All your observations are correct, your are not doing anything wrong, but it isn't exactly a bug either.

`abc -mux8` and `abc -mux16` give all the information to ABC needed in theory to extract those cell types, but then ABC doesn't do it, I think because it's ignoring cells above a certain size. That's also why the `-mux<N>` options are commented out in the `yosys -h abc` help message.

SergeyDegtyar, 2019-05-02:
Ok. Thank you.

closed 2019-05-02 by SergeyDegtyar