YosysHQ/yosys #920: synth_easic reports an error about missing files.

SergeyDegtyar, 2019-04-08:
Automated test:
https://github.com/YosysHQ/yosys-tests/tree/master/architecture/synth_easic

## Steps to reproduce the issue
top.v:
module adff
    ( input d, clk, clr, output reg q );
    initial begin
      q = 0;
    end
	always @( posedge clk, posedge clr )
		if ( clr )
			q <= 1'b0;
		else
            q <= d;
endmodule

script:
read_verilog top.v
synth_easic

## Expected behavior

Synthesis finished.

## Actual behavior

yosys> synth_easic

8. Executing SYNTH_EASIC pass.

8.1. Executing Liberty frontend.
ERROR: Can't open input file `/opt/eTools/data_ruby28/design_libs/logical/timing/gp/n3x_phys_clk_0v893ff125c.lib' for reading: No such file or directory

Is it bug or I need install and use some additional tool for this command?

cliffordwolf, 2019-05-01:
> Is it bug or I need install and use some additional tool for this command?

Yes, not a bug, this requires the easic "etools" package in /opt/eTools/. See also `help synth_easic`:

```
    -etools <path>
        set path to the eTools installation. (default=/opt/eTools)
```

closed 2019-05-01 by cliffordwolf

jalcim, 2022-04-03:
> > Is it bug or I need install and use some additional tool for this command?
> 
> Yes, not a bug, this requires the easic "etools" package in /opt/eTools/. See also `help synth_easic`:
> 
> ```
>     -etools <path>
>         set path to the eTools installation. (default=/opt/eTools)
> ```

Maybe can you precise what is eTools ?
Or when can we find documentation.

Thanks a lot.