YosysHQ/yosys #921: Can't simulate synthesized for intel/cyclonev design in iverilog.

SergeyDegtyar, 2019-04-08:
Automated test case:
https://github.com/YosysHQ/yosys-tests/tree/master/architecture/synth_intel_cyclonev/

## Steps to reproduce the issue
testbench.v
module testbench;
    reg [2:0] in;

	wire patt_out,out;
	wire patt_carry_out,carryout;

    initial begin
        // $dumpfile("testbench.vcd");
        // $dumpvars(0, testbench);

        #5 in = 0;
        repeat (10000) begin
            #5 in = in + 1;
        end

        $display("OKAY");
    end

    top uut (
	.x(in[0]),
	.y(in[1]),
	.cin(in[2]),
	.A(out),
	.cout(carryout)
	);
	
	assign {patt_carry_out,patt_out} =  in[2] + in[1] + in[0];
endmodule

top.v:
module top
(
 input x,
 input y,
 input cin,

 output A,
 output cout
 );

`ifndef BUG
assign {cout,A} =  cin + y + x;
`else
assign {cout,A} =  cin - y * x;
`endif

endmodule

script for yosys:
read_verilog ../top.v
synth_intel -family cyclonev
write_verilog synth.v 

command for iverilog:
iverilog -o testbench  ../testbench.v synth.v ../../common.v ../../../../../techlibs/common/simcells.v ../../../../../techlibs/intel/cyclonev/cells_sim.v

## Expected behavior

Simulation finished.

## Actual behavior

+ iverilog -o testbench ../testbench.v synth.v ../../common.v ../../../../../techlibs/common/simcells.v ../../../../../techlibs/intel/cyclonev/cells_sim.v
../../../../../techlibs/intel/cyclonev/cells_sim.v:88: error: Unable to bind wire/reg/memory `upper_mask_value' in `testbench.uut._05_.lut5'
../../../../../techlibs/intel/cyclonev/cells_sim.v:101: error: Unable to bind wire/reg/memory `upper_mask_value' in `testbench.uut._05_.lut6'
../../../../../techlibs/intel/cyclonev/cells_sim.v:111: error: shared_lut_alm is not a valid l-value in testbench.uut._05_.
../../../../../techlibs/intel/cyclonev/cells_sim.v:62:      : shared_lut_alm is declared here as wire.
../../../../../techlibs/intel/cyclonev/cells_sim.v:113: error: Could not find variable ``out_0'' in ``testbench.uut._05_''
../../../../../techlibs/intel/cyclonev/cells_sim.v:114: error: Could not find variable ``out_1'' in ``testbench.uut._05_''
../../../../../techlibs/intel/cyclonev/cells_sim.v:115: error: Could not find variable ``out_2'' in ``testbench.uut._05_''
../../../../../techlibs/intel/cyclonev/cells_sim.v:116: error: Could not find variable ``out_3'' in ``testbench.uut._05_''
../../../../../techlibs/intel/cyclonev/cells_sim.v:107: warning: @* found no sensitivities so it will never trigger.
../../../../../techlibs/intel/cyclonev/cells_sim.v:88: error: Unable to bind wire/reg/memory `upper_mask_value' in `testbench.uut._06_.lut5'
../../../../../techlibs/intel/cyclonev/cells_sim.v:101: error: Unable to bind wire/reg/memory `upper_mask_value' in `testbench.uut._06_.lut6'
../../../../../techlibs/intel/cyclonev/cells_sim.v:111: error: shared_lut_alm is not a valid l-value in testbench.uut._06_.
../../../../../techlibs/intel/cyclonev/cells_sim.v:62:      : shared_lut_alm is declared here as wire.
../../../../../techlibs/intel/cyclonev/cells_sim.v:113: error: Could not find variable ``out_0'' in ``testbench.uut._06_''
../../../../../techlibs/intel/cyclonev/cells_sim.v:114: error: Could not find variable ``out_1'' in ``testbench.uut._06_''
../../../../../techlibs/intel/cyclonev/cells_sim.v:115: error: Could not find variable ``out_2'' in ``testbench.uut._06_''
../../../../../techlibs/intel/cyclonev/cells_sim.v:116: error: Could not find variable ``out_3'' in ``testbench.uut._06_''
../../../../../techlibs/intel/cyclonev/cells_sim.v:107: warning: @* found no sensitivities so it will never trigger.
14 error(s) during elaboration.
make: *** [Makefile:53: synth_intel_cyclonev/work_synth_intel_cyclonev/.stamp] Error 14

dh73, 2019-04-11:
The issue is quite obviuos, I'm sorry you found it. I've already fixed on my branch. Will make a PR in the morning (America timezone).
Also, I will not suggest use write_verilog command to generate the VQM or you'll be warranted to have problems with Quartus.
Better use
```bash
synth_intel - family cyclonev -vqm  synth.v
```

dh73, 2019-04-12:
PR in progress:
https://github.com/YosysHQ/yosys/pull/933#issue-269838634

After this change is merged, you will be able to see your expected result:
```bash
./testbench                                                                                   
OKAY
```

mmicko, 2019-04-12:
@dh73 while you are here, can you please confirm if https://github.com/YosysHQ/yosys/pull/911 is fine ? (it is for gowin so sorry for taking over the thread :) )

cliffordwolf, 2019-05-01:
Closing under the assumption that #933 fixed this.

closed 2019-05-01 by cliffordwolf