YosysHQ/yosys #922: Can't get write_firrtl output for design than contains memory.

SergeyDegtyar, 2019-04-08:
In all cases what I tried to use I had this error in the report:  

> ERROR: Unclocked write port 0 on memory top.ram. 

Do I need some special RAM description for write_firrtl or it is bug?

## Steps to reproduce the issue
top.v:
```
module top
(
	input [7:0] data_a,
	input [6:1] addr_a,
	input we_a, clk,
	output reg [7:0] q_a
);
	// Declare the RAM variable
	reg [7:0] ram[63:0];

	// Port A
	always @ (posedge clk)
	begin
		if (we_a)
		begin
			ram[addr_a] <= data_a;
			q_a <= data_a;
		end
			q_a <= ram[addr_a];
	end

endmodule
```
script:
read_verilog ../top.v
memory_collect
proc
write_firrtl f.f

## Expected behavior

Firrtl backend successfully executed.

## Actual behavior

yosys> write_firrtl f.f

4. Executing FIRRTL backend.

4.1. Executing PMUXTREE pass.
Warning: Initial value (00000000) for (top.q_a) not supported
Warning: Initial value (00000000) for (top.q_b) not supported
ERROR: Unclocked write port 0 on memory top.ram.

daveshah1, 2019-04-12:
The unclocked write port, which the FIRRTL backend doesn't support (quite possibly because FIRRTL doesn't support them) is because you aren't running `memory_dff` (in this case `memory_dff -nordff` is needed as the FIRRTL backed doesn't support _clocked_ read ports either).

Finally, I find that `opt_reduce; clean` is needed to simplify the write enable expression sufficiently for the FIRRTL backend to work. I also reordered things so that `proc` was run first to extract DFFs before any memory passes.

The script that works for me is
```
proc
memory_dff -nordff
memory_collect
opt_reduce
clean
write_firrtl f.f
```

SergeyDegtyar, 2019-04-15:
Thank you! I have checked. Your script works.

closed 2019-04-15 by SergeyDegtyar