YosysHQ/yosys #924: Internal error in internal cell is found in case when use "insbuf -buf" command

SergeyDegtyar, 2019-04-08:
Automated test case:
https://github.com/YosysHQ/yosys-tests/tree/master/misc/insbuf

## Steps to reproduce the issue

test case:
```
module top    (
out,
i,
clk,
o,
in
);
    output [7:0] out;
    input clk, in;
    reg [7:0] out;
    input i;
    output o;

    always @(posedge clk)
	begin
		out    <= out << 1;
		out[0] <= in;
	end

endmodule
```
script:
read_verilog ../top.v
synth
insbuf -buf $_BUF_ in out

## Expected behavior

Command successfully executed.

## Actual behavior

yosys> insbuf -buf $_BUF_ in out

3. Executing INSBUF pass (insert buffer cells for connected wires).
Added top.$auto$insbuf.cc:79:execute$57: 1'x -> \o
ERROR: Found error in internal cell \top.$auto$insbuf.cc:79:execute$57 ($_BUF_) at kernel/rtlil.cc:772:
  cell $_BUF_ $auto$insbuf.cc:79:execute$57
    connect \out \o
    connect \in 1'x
  end

daveshah1, 2019-04-12:
This seems like expected behaviour. You are creating cells called `$_BUF_` which is a Yosys internal cell (see https://github.com/YosysHQ/yosys/blob/master/techlibs/common/simcells.v#L28-L43) and creating ports `in` and `out` on this cell. But the correct names for the ports on a `$_BUF_` cell are `A` and `Y`, therefore this is an internal cell error (a Yosys internal cell type with incorrect port naming).

SergeyDegtyar, 2019-04-15:
Ok. Thank you. Then I will fix the test for this case.

closed 2019-04-15 by SergeyDegtyar