YosysHQ/yosys #925: Unexpected error is reported in case when use 'bugpoint' command.

SergeyDegtyar, 2019-04-08:
## Steps to reproduce the issue

Automated test case:
https://github.com/YosysHQ/yosys-tests/tree/master/misc/bugpoint

script:
read_verilog ../top.v
bugpoint

## Expected behavior

Executed successfully

## Actual behavior

+ cd bugpoint/work_bugpoint
+ yosys -ql yosys.log ../../scripts/bugpoint.ys
ERROR: No such command: autoidx (type 'help' for a command overview)
ERROR: No such command: autoidx (type 'help' for a command overview)
ERROR: No such command: autoidx (type 'help' for a command overview)

daveshah1, 2019-04-12:
The bugpoint pass needs a script to be passed to it, otherwise it is pointless (the purpose is to minimise a design until the minimum design that causes a crash with a given script), the help says:
```
    -script <filename>
        use this script to crash Yosys. required.
```
Nonetheless, I'm tagging this as a bug because we should produce a nice error message when a script isn't given.

cliffordwolf, 2019-05-01:
cc @whitequark

PR #1007 (2019-05-14): bugpoint: check for -script option
Fixes #925.
commit c8c1df23 (2019-05-14): bugpoint: check for -script option.

Fixes #925.
closed 2019-05-14 by whitequark