YosysHQ/yosys #926: Internal error in internal cell is found in case when use "setparam -type" command

SergeyDegtyar, 2019-04-08:
#Automated test case:
https://github.com/YosysHQ/yosys-tests/tree/master/misc/setparam

## Steps to reproduce the issue

test case:
```
module top
(
 input x,
 input y,
 input cin,

 output reg A,
 output cout
 );
 parameter WIDTH = 1;
 wire o;

`ifndef BUG
always @(posedge cin)
	A <= o;

assign cout =  cin? y : x;

middle u_mid (x,y,o);
`else
assign {cout,A} =  cin - y * x;
`endif

endmodule

module middle
(
	input x,
	input y,
	output o
);

assign o = x + y;
endmodule
```
script:
read_verilog ../top.v
proc
setparam -type $mux -set WIDTH 1 top

## Expected behavior

Command successfully executed.

## Actual behavior

yosys> setparam -type $mux -set WIDTH 1
ERROR: Found error in internal cell \middle.$add$top.v:33$3 ($mux) at kernel/rtlil.cc:741:
  attribute \src "top.v:33"
  cell $mux $add$top.v:33$3
    parameter \WIDTH 1
    parameter \Y_WIDTH 1
    parameter \B_WIDTH 1
    parameter \A_WIDTH 1
    parameter \B_SIGNED 0
    parameter \A_SIGNED 0
    connect \Y $add$top.v:33$3_Y
    connect \B \y
    connect \A \x
  end

daveshah1, 2019-04-12:
The command `setparam -type $mux -set WIDTH 1` will set all cells to type `$mux` and also set parameter `WIDTH` to 1. But `$mux` is a Yosys internal cell type with a specific interface, and setting other cells (in this case what appears to have been an `$add` cell) to `$mux` type will mean they don't meet that interface (e.g. in this case there is no `S` input). So this error seems correct.

SergeyDegtyar, 2019-04-15:
Ok. Thank you. Will fix the test.

closed 2019-04-15 by SergeyDegtyar