YosysHQ/yosys #930: Optimise $mul followed by $div

eddiehung, 2019-04-11:
## Steps to reproduce the issue

`./yosys -p 'prep; opt -full -fine; stat` 

```
module top(input [1:0] i, output [1:0] o);
wire [3:0] t;
assign t = i * 3;
assign o = t / 3;
endmodule
```

## Expected behavior

Should boil down to `assign o = i;`

## Actual behavior

`$mul` cell followed by `$div` cell: [show.pdf](https://github.com/YosysHQ/yosys/files/3069715/show.pdf)

janrinze, 2019-04-27:
Should check if the intermediate result did not overflow or otherwise got truncated.

cliffordwolf, 2019-04-30:
PR #969 adds a peephole optimization for this

cliffordwolf, 2019-04-30:
link to pmgen file:
https://github.com/YosysHQ/yosys/pull/969/files#diff-38f18033e8f2b0e3295a7185e4316beb

commit b515fd2d (2019-04-30): Add peepopt_muldiv, fixes #930
closed 2019-05-03 by cliffordwolf