YosysHQ/yosys #931: Full case statement with default entry creates $dlatch

eddiehung, 2019-04-11:
## Steps to reproduce the issue

`yosys -p "prep" case.v`
```
module mux_case_4_1 (input [4-1:0] i, input [2-1:0] s, output reg [0:0] o);
always @*
    case (s)
        0: o <= i[0];
        1: o <= i[1];
        2: o <= i[2];
        3: o <= i[3];
        default: o <= 1'bx;
    endcase
endmodule
```

## Expected behavior

```
   Number of cells:                  5
     $eq                             3
     $logic_not                      1
     $pmux                           1
```

## Actual behavior

```
   Number of cells:                 14
     $and                            3
     $dlatch                         1
     $eq                             3
     $logic_not                      1
     $not                            5
     $pmux                           1
```

This works as expected:
```
module mux_case_4_1 (input [4-1:0] i, input [2-1:0] s, output reg [0:0] o);
always @*
    case (s)
        //0: o <= i[0];
        1: o <= i[1];
        2: o <= i[2];
        3: o <= i[3];
        default: o <= i[0];
    endcase
endmodule
```

eddiehung, 2019-04-11:
More fun: changing the non blocking assignments to blocking fixes it too...

eddiehung, 2019-04-16:
Seems that #932 does not really fix this problem, reverting because it broke #935.

eddiehung, 2019-04-16:
@cliffordwolf looks like I'll need some help from you.

It appears that the following circuits, with blocking `=` assignments, do not create a latch, however, non blocking assignments `<=` do create latches:
```
module mux_case_4_1 (input [4-1:0] i, input [2-1:0] s, output reg [1:0] o);
always @*
    case (s)
        0: o[0] = i[0];
        1: o[0] = i[1];
        2: o[0] = i[2];
        3: o[0] = i[3];
        default: o[0] = i[0];
    endcase
endmodule
```
```
module mux_case_4_1 (input [4-1:0] i, input [2-1:0] s, output reg [1:0] o);
always @*
    case (s)
        0: o[0] = i[0];
        1: o[0] = i[1];
        2: o[0] = i[2];
        3: o[0] = i[3];
        //default: o[0] = i[0];
    endcase
endmodule
```
```
module mux_case_4_1 (input [4-1:0] i, input [2-1:0] s, output reg [1:0] o);
always @*
    (* full_case *)
    case (s)
        0: o[0] = i[0];
        1: o[0] = i[1];
        2: o[0] = i[2];
        //3: o[0] = i[3];
        //default: o[0] = i[0];
    endcase
endmodule
```

eddiehung, 2019-04-16:
My current approach has been to change `proc_rmdead` so that it correctly recognises default statements, even when they all values are specified, but that doesn't capture all cases, on top of breaking other things.
Other approaches I've been looking at is to change `proc_mux` or `proc_dlatch`...

cliffordwolf, 2019-04-18:
I think #942 should fix this issue. The following produces the expected results for me. (The `default` case in the examples above have been optimized away anyway because `proc_rmdead` is smart enough to understand that they are unreachable.)

```
module test (
        input [1:0] S,
        input [7:0] A, B, C, D,
        output reg [7:0] Y
);
        always @* begin
                case (S)
                        2'b00: Y = A;
                        2'b01: Y = B;
                        2'b10: Y = C;
                        2'b11: Y = D;
                endcase
        end
endmodule
```

PR #942 (2019-04-20): Improve proc full_case detection and handling
this should fix #931
commit 88be1cbf (2019-04-20): Improve proc full_case detection and handling, fixes #931
closed 2019-04-20 by cliffordwolf