YosysHQ/yosys #935: Logic loopo

anishathalye, 2019-04-13:
## Steps to reproduce the issue

This issue occurs with 7685469e (found via bisect) as well as current master (db1a5ec6), but not in 0deaccba.

I'm not sure if this is an issue with yosys, picorv32, or a lack of my own knowledge about these tools (if it's the last one, apologies, feel free to close the issue).

I tried running the following on `picorv32.v` (from https://github.com/cliffordwolf/picorv32, latest master = f48f5fe9704fd34bf3de39dbced427ea2e2d8895):

```
yosys -p 'read_verilog picorv32.v' -p 'prep -top picorv32 -nordff' -p 'opt -fast' -p 'write_smt2 picorv32.smt2
```

## Expected behavior

With the older version of yosys, it works and generates the .smt2 file.

## Actual behavior

On 7685469e onwards, the conversion fails with:

```
...
-- Running command `write_smt2 picorv32.smt2' --

4. Executing SMT2 backend.
ERROR: Found logic loop in module picorv32! See cell $auto$proc_mux.cc:375:signal_to_mux_tree$3581.
```

(full log: https://gist.github.com/anishathalye/a754d478c91615edbf02f983f2beb184)

eddiehung, 2019-04-13:
Hi @anishathalye, thanks for such a detailed report. This is definitely an unexpected result, so I'll take a look!

eddiehung, 2019-04-16:
It looks like #931 -- which is meant to eliminate latches in certain places -- is actually causing latches to appear in other places...! I'm reverting it for now.

eddiehung, 2019-04-16:
I'm AFK but #937 which reverts #932 has now been merged which should bring everything back to working. Sorry for the regression!

anishathalye, 2019-04-26:
Yes, it works for me too. Thanks for looking into it!

closed 2019-04-26 by anishathalye