YosysHQ/yosys #938: Crash occurs in case when use write_firrtl command for design with $memwr and $memrd cells.

SergeyDegtyar, 2019-04-16:
Automated test case:
https://github.com/YosysHQ/yosys-tests/tree/master/backends/write_firrtl_mem

In case when I had $memwr cell in the design I had this report:  
terminate called after throwing an instance of 'std::out_of_range'
  what():  dict::at()
Aborted (core dumped)

The difference from #922 is that I don't use memory_collect command here.

## Steps to reproduce the issue
top.v:
```
module top
(
	input [7:0] data_a,
	input [6:1] addr_a,
	input we_a, clk,
	output reg [7:0] q_a
);
	// Declare the RAM variable
	reg [7:0] ram[63:0];

	// Port A
	always @ (posedge clk)
	begin
		if (we_a)
		begin
			ram[addr_a] <= data_a;
			q_a <= data_a;
		end
			q_a <= ram[addr_a];
	end

endmodule
```
script:
read_verilog ../top.v
proc
memory_dff -nordff
opt_reduce
clean
write_firrtl firrtl.firrtl

## Expected behavior

Firrtl backend successfully executed.

## Actual behavior

yosys> write_firrtl f.f

4. Executing FIRRTL backend.

4.1. Executing PMUXTREE pass.
yosys> write_firrtl f.f

5. Executing FIRRTL backend.

5.1. Executing PMUXTREE pass.
terminate called after throwing an instance of 'std::out_of_range'
  what():  dict::at()
Aborted (core dumped)

cliffordwolf, 2019-04-30:
CC @ucbjrl

ucbjrl, 2019-04-30:
Thanks @SergeyDegtyar for the simple test case. I hope to have a fix in the very near future.

PR #977 (2019-05-01): Fix #938 - Crash occurs in case when use write_firrtl command
Add missing memory initialization.
Sanity-check memory parameters.
Add Cell pointer to memory object (for error reporting).

This should resolve #938
commit 38f5424f (2019-05-01): Fix #938 - Crash occurs in case when use write_firrtl command
Add missing memory initialization.
Sanity-check memory parameters.
Add Cell pointer to memory object (for error reporting).
commit e8a157b4 (2019-05-01): Merge pull request #977 from ucb-bar/fixfirrtlmem

Fix #938 - Crash occurs in case when use write_firrtl command
closed 2019-05-01 by cliffordwolf