YosysHQ/yosys #940: Assertion error in abc

phheim, 2019-04-17:
When yosys is called with "synth_ice40" to generate a json file, it sometimes terminates with

> ...
3.39.1.1. Executing ABC.
Running ABC command: <yosys-exe-dir>/yosys-abc -s -f <abc-temp-dir>/abc.script 2>&1
ABC: ABC command line: "source <abc-temp-dir>/abc.script".
ABC: 
ABC: + read_blif <abc-temp-dir>/input.blif 
ABC: + read_lut <abc-temp-dir>/lutdefs.txt 
ABC: + strash 
ABC: + ifraig 
ABC: + scorr 
ABC: Warning: The network is combinational (run "fraig" or "fraig_sweep").
ABC: + dc2 
ABC: + dretime 
ABC: + strash 
ABC: + dch -f 
ABC: + if 
ABC: + mfs2 
ABC: + lutpack -S 1 
ABC: + dress 
ABC: Total number of equiv classes                =    2336.
ABC: Participating nodes from both networks       =    5529.
ABC: Participating nodes from the first network   =    2451. (  84.69 % of nodes)
ABC: Participating nodes from the second network  =    3078. ( 106.36 % of nodes)
ABC: Node pairs (any polarity)                    =    2450. (  84.66 % of names can be moved)
ABC: Node pairs (same polarity)                   =    2142. (  74.02 % of names can be moved)
ABC: Total runtime =     0.27 sec
ABC: + write_blif <abc-temp-dir>/output.blif 
ABC: yosys-abc: src/base/abc/abcFanio.c:92: void Abc_ObjAddFanin(Abc_Obj_t *, Abc_Obj_t *): Assertion `!Abc_ObjIsNet(pObj) || !Abc_ObjFaninNum(pObj)' failed.
ABC: Aborted (core dumped)
ERROR: ABC: execution of command "/home/work/tools/yosys/yosys-abc -s -f /tmp/yosys-abc-efsP93/abc.script 2>&1" failed: return code 134.

daveshah1, 2019-04-17:
Please provide your input Verilog, otherwise it will be impossible to reproduce this issue.

phheim, 2019-04-17:
Sorry I forgot. The command used to generate this error is 
> yosys  -p "synth_ice40 -top SuperTopEntity -json TopEntity.json" *.v

[source.zip](https://github.com/YosysHQ/yosys/files/3088925/source.zip)

daveshah1, 2019-04-19:
Thanks for the report!

This has been fixed upstream in ABC, and Yosys' ABC revision updated in e6253244899760e580fd822f3017259f8f15a45c

closed 2019-04-19 by daveshah1