YosysHQ/yosys #948: Valid RTLIL produces invalid Verilog, which Yosys then cannot read back

Wren6991, 2019-04-22:
From [an nMigen issue](https://github.com/m-labs/nmigen/issues/41).

## Steps to reproduce the issue
bug.rtlil:
```
module \top
  wire width 1 input 0 $signal
  wire width 1 output 1 \a
  wire width 1 $next\a
  wire width 1 $1
  cell $reduce_bool $2
    parameter \A_SIGNED 0
    parameter \A_WIDTH 0
    parameter \Y_WIDTH 1
    connect \A {}
    connect \Y $1
  end
  process $group_0
    assign $next\a 1'0
    assign $next\a $1
    sync init
    sync always
      update \a $next\a
  end
end
```

On latest master (0e7901e4), run the following:

```
yosys -p "read_ilang bug.rtlil; write_verilog bug.v; delete; read_verilog bug.v"
```

Yosys gives a parse error when reading back the Verilog it has just generated. The cause is a zero-item concatenation list, which is *syntactically* illegal (1364-2005 A.8.1).

However, the following succeeds:

```
yosys -p "read_ilang bug.rtlil; synth; write_verilog bug.v; delete; read_verilog bug.v"
```
This produces a `1'h0` assignment to `a`.

## Expected behavior

Zero-item concatenations are explicitly special-cased in `ilang_parser.y`, and zero-width wires are not disallowed in the RTLIL documentation. As far as I have read through the source code (admittedly not very) Yosys copes with these in its in-memory representation.

OTOH the Verilog standard explicitly forbids zero-item concatenations, forbids zero-length `wire`s in a roundabout way, and is surprisingly woolly on zero-length signals produced by expressions (with surprising edge cases, e.g. being allowed inside of concatenations).

As:
- Zero-length signals are not disallowed by RTLIL, but are disallowed by Verilog
- The or reduction of a zero-width signal is logically zero (empty sum convention)
- The current synthesis behaviour is to produce zero for the or reduction

I would expect `read_ilang bug.rtlil; write_verilog bug.v` to produce a zero assignment to `a`. If not, some other valid Verilog, based on some documented behaviour. Note that the above is a minimal example to provoke the problematic behaviour, but it was isolated from real code.

## Actual behavior

Yosys produces invalid Verilog, which it cannot then read back. Paste of `write_verilog` output:

```
/* Generated by Yosys 0.8+354 (git sha1 0e7901e4, gcc 8.2.0-7ubuntu1 -fPIC -O2) */

module top(a, _2_);
  wire _0_;
  reg _1_;
  input _2_;
  output a;
  reg a;
  assign _0_ = | {  };  // syntax error here
  always @* begin
    _1_ = 1'h0;
    _1_ = _0_;
  end
  initial begin
  end
  always @* begin
      a <= _1_;
  end
endmodule
```

cliffordwolf, 2019-04-22:
1) You should never use `write_verilog` with an RTLIL design with processes. It even gives you the following warning:

```
Warning: Module top contains unmapped RTLIL processes. RTLIL processes
can't always be mapped directly to Verilog always blocks. Unintended
changes in simulation behavior are possible! Use "proc" to convert
processes to logic networks and registers.
```

2) I think I've fixed it in #953. But generating Verilog with zero-width expressions is really asking for troubles imho. I think it's better to get rid of them by running something like `opt_expr` before writing the Verilog output.

whitequark, 2019-04-23:
> 1. You should never use `write_verilog` with an RTLIL design with processes. It even gives you the following warning:

This is true in general, but nMigen specifically generates only RTLIL processes that it knows it can use with `write_verilog`. The alternative to this is not using Yosys or RTLIL at all, which I think you will agree is a massive waste of time.

cliffordwolf, 2019-04-28:
>  The alternative to this is not using Yosys or RTLIL at all, which I think you will agree is a massive waste of time.

Running `proc` is not an option?

whitequark, 2019-04-28:
> Running `proc` is not an option?

The output of oMigen has historically been readable, since it is the source of truth that is used when debugging incorrect behavior of the kind that is not easily visible from Python. I've added tooling in nMigen to handle the most common cases without having to resort to read the processed output but it is not a silver bullet.

Althuogh it has been readable, it has never been particularly easy to read, especially due to the fact that the design was completely flat. nMigen has fixed this and produces hierarchical designs now. But reading RTLIL corresponding for them almost negates that advantage due to its verbosity; RTLIL is a reasonably good IR but it is not so useful as an output of a tool *for finding problems at a glance*.

That's the primary reason nMigen emits Verilog, and that's why nMigen emits Verilog with processes, and of course that's the reason for having #726. If I am not able to use Yosys to do so, I would have to implement a Verilog backend for nMigen from scratch; and, the Verilog backend in oMigen is seriously buggy in all the little nasty Verilog edge cases Yosys is good at handling, and really should be the single source of truth for, in this case.

whitequark, 2019-07-14:
Reopening since it seems unclear per LRM whether `""` has width 0 or 8.

commit 0e0c80fa (2019-04-22): Add support for zero-width signals to Verilog back-end, fixes #948
PR #953 (2019-04-22): Add support for zero-width signals to Verilog back-end
Fixes #948
PR #1203 (2019-07-18): write_verilog: dump zero width constants correctly
Before this commit, zero width constants were dumped as "" (empty
string). Unfortunately, 1364-2005 5.2.3.3 indicates that an empty
string is equivalent to "\0", and is 8 bits wide, so that's wrong.

After this commit, a replication operation with a count of zero is
used instead, which is explicitly permitted per 1364-2005 5.1.14,
and is defined to have size zero. (Its operand has to have a non-zero
size for it to be legal, though.)

Fixes #948 (again).
commit 4ff44d85 (2019-07-18): write_verilog: dump zero width constants correctly.

Before this commit, zero width constants were dumped as "" (empty
string). Unfortunately, 1364-2005 5.2.3.3 indicates that an empty
string is equivalent to "\0", and is 8 bits wide, so that's wrong.

After this commit, a replication operation with a count of zero is
used instead, which is explicitly permitted per 1364-2005 5.1.14,
and is defined to have size zero. (Its operand has to have a non-zero
size for it to be legal, though.)

Fixes #948 (again).
closed 2019-07-18 by cliffordwolf