YosysHQ/yosys #956: unprocessed 'init' attribute

ZipCPU, 2019-04-23:
## Steps to reproduce the issue

Consider the following code:
```
module inivalue(i_clk, i_val, o_val);
	input	wire	i_clk;
	input	wire	[2:0]	i_val;
	output	reg	[1:0]	o_val;

	reg	[2:0]	r_val;

	initial	r_val = 0;
	always @(posedge i_clk)
		r_val <= r_val + i_val;

	always @(*)
		o_val = r_val[1:0];

endmodule
```

Notice the unused bits of r_val.

If you process this code with either `yosys -p 'synth_ice40 -top inivalue' inivalue.v` or `yosys -p 'synth_ecp5 -top inivalue' inivalue.v`, yosys will return the warning:

```
Warning: Wire inivalue.r_val has an unprocessed 'init' attribute.
```

This warning suggests that an `initial` attribute has not been applied, when in reality what's taking place appears to be an appropriate optimization.

## Expected behavior

I expect either 1) no warnings if the synthesized design matches the incoming warning, or 2) a more specific warning indicating that the unprocessed init attribute is associated with an unused bit.

## Actual behavior

See the warning above.

eddiehung, 2019-04-25:
If it helps at all, I'm seeing messages like this in the yosys-bench CI (running `synth_ice40`) too: https://cirrus-ci.com/task/6520543497945088

cliffordwolf, 2019-04-29:
I think #966 fixes this. @eddiehung can you confirm that this also fixes the warnings you are seeing in yosys-bench?

ZipCPU, 2019-04-29:
This seems to fix the issue for the MCVE listed above, but not in general.  I'm still getting several other errors of this same type.

Let me see if I can put together another MCVE to help chase down the other examples I have of this bug.

ZipCPU, 2019-04-29:
I have three remaining examples of this, and each of the three examples I have crosses logic between large modules, so generating the MCVE takes some amount of work.  That said, I managed to create a second one, attached below.  Strangely, this error only occurs with synth_ecp5, and not with synth_ice40

[sixremain.zip](https://github.com/YosysHQ/yosys/files/3128228/sixremain.zip)

cliffordwolf, 2019-04-30:
Pushed another commit to #966. Please let me know if that fixes things. @ZipCPU

ZipCPU, 2019-04-30:
This fixes all of my problems with these initial warnings.  Thank you.  Feel free to close this issue when you merge #966 .

eddiehung, 2019-05-02:
Negatory from me. I'm still seeing:

```
yosys -qp synth_ice40 sddac.v
Warning: Wire sddac.quant_in has an unprocessed 'init' attribute.
Warning: Wire sddac.quantizer has an unprocessed 'init' attribute.
```

when applied to sddac from [here](https://raw.githubusercontent.com/YosysHQ/yosys-bench/master/verilog/benchmarks_large/sddac/sddac.v).

cliffordwolf, 2019-05-03:
> Warning: Wire sddac.quant_in has an unprocessed 'init' attribute.
> Warning: Wire sddac.quantizer has an unprocessed 'init' attribute.

@eddiehung These warnings are both completely justified afaict. Both of those regs are initialized to 0 but are then driven by an `always @*` block.

```
    reg signed [20:0] quant_in = 0;     // Q(2,19)
...
    reg               quantizer = 0;
...
    always @(*)
    begin
...
        quant_in  <= state2 + $signed(lfsr_fb ? -21'h4000 : 21'h4000);
...
        quantizer <= quant_in[20];
    end
```

And the wires are being used, so no no-warnings-exception for unused regs.

CC @trcwm (Also, using non-blocking assignments (`<=`) in an `always @*` can lead to race conditions in simulation. http://www.sunburst-design.com/papers/CummingsSNUG2000SJ_NBA_rev1_2.pdf It's not as bad as using blocking assignments incorrectly in a clocked always block, but still super hard to troubleshoot if you get unlucky enough to encounter such a race condition in the wild.)

trcwm, 2019-05-03:
Init and non-blocking assignment issues in SDDAC should be fixed now.

cliffordwolf, 2019-05-03:
Thanks @trcwm !

@eddiehung Any other suspicious warnings of this kind that I should have a look at?

eddiehung, 2019-05-03:
All good from what I can tell.

For the record, these files no longer produce an init warning: 
https://raw.githubusercontent.com/alexforencich/verilog-ethernet/fb4abb6b39fe1536162dfdc41d3cf64beb8fa834/rtl/ip_eth_tx_64.v
https://raw.githubusercontent.com/alexforencich/verilog-ethernet/8b8cfd96fdc95913c87ba42188c5045f340df6b0/lib/axis/rtl/axis_fifo.v

commit 754b1ee4 (2019-04-29): Drive dangling wires with init attr with their init value, fixes #956
PR #966 (2019-04-30): Drive dangling wires with init attr with their init value
Fixes #956.
closed 2019-05-03 by cliffordwolf